carla-simulator/carla #3167: CARLA's C++ API error

meitalcohenyaacobi, 2020-08-05:
Hi,
I'm trying to compile the example which creates an application using CARLA's C++ API.
while running >> make run
I got the following errors:
[1/99] Building CXX object LibCarla/cmake/client/CMakeFiles/carla_client.dir/__/__/source/carla/StringUtil.cpp.o
FAILED: LibCarla/cmake/client/CMakeFiles/carla_client.dir/__/__/source/carla/StringUtil.cpp.o 
/usr/bin/g++-7  -DBOOST_ERROR_CODE_HEADER_ONLY -DLIBCARLA_IMAGE_WITH_PNG_SUPPORT=true -DLIBCARLA_TEST_CONTENT_FOLDER=\"/mnt/workdisk/carlaTest/carla/Build/test-content\" -I../../../LibCarla/cmake/../source -I../../../LibCarla/cmake/../source/third-party -isystem ../../../Build/boost-1.72.0-c8-install/include -isystem ../../../Build/rpclib-v2.2.1_c3-c8-libstdcxx-install/include -isystem ../../../Build/recast-cdce4e-c8-install/include -std=c++14 -pthread -fPIC -O3 -DNDEBUG -Werror -Wall -Wextra -std=c++14 -pthread -fPIC -O3 -DNDEBUG -Werror -Wall -Wextra   -O3 -DNDEBUG -MD -MT LibCarla/cmake/client/CMakeFiles/carla_client.dir/__/__/source/carla/StringUtil.cpp.o -MF LibCarla/cmake/client/CMakeFiles/carla_client.dir/__/__/source/carla/StringUtil.cpp.o.d -o LibCarla/cmake/client/CMakeFiles/carla_client.dir/__/__/source/carla/StringUtil.cpp.o -c ../../../LibCarla/source/carla/StringUtil.cpp
In file included from ../../../LibCarla/source/carla/StringUtil.cpp:7:0:
../../../LibCarla/cmake/../source/carla/StringUtil.h:9:10: fatal error: boost/algorithm/string.hpp: No such file or directory
 #include <boost/algorithm/string.hpp>
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~

germanros1987, 2020-08-19:
Sorry for the delay, we are in the middle of the vacation season. @Axel1092 could you please try to help with this?

Axel1092, 2020-08-20:
Hi @meitalcohenyaacobi, what example are you trying to compile? I think you are using g++ instead of clang which is the supported compiler.

meitalcohenyaacobi, 2020-08-20:
Hi,
I used this page https://github.com/carla-simulator/carla/tree/master/Examples/CppClient
in order to download and run the cpp client.

Axel1092, 2020-08-26:
Hi @meitalcohenyaacobi, maybe you didn't compile Carla beforehand. Try running `make LibCarla` before the examples so that it compiles Carla and other necessary dependencies.

meitalcohenyaacobi, 2020-08-27:
Hi,
I did run `make LibCarla` before,
Are there any necessary dependencies in addition to to LibCarla?
Thanks.

Axel1092, 2020-08-27:
The error shows that there is a problem with boost, maybe it has not been properly downloaded during the make command. Try to remove any boost files and folders from the Build folder in the carla directory, then run `make LibCarla` again and then the examples.

closed 2021-04-27 by stale[bot]