carla-simulator/carla #3168: Vehicle Max Acceleration

lucasnmatias, 2020-08-06:
Hi, I work at IVEX and our focus is on the evaluation of autonomous vehicles behavior and decision making. We use CARLA to implement challenging scenarios to evaluate vehicles behavior in those scenarios. Right we are struggling to control the vehicles acceleration.

Here is my question, is there any easy way to limit the max acceleration of a vehicle in CARLA?

Glazyee, 2020-08-06:
I have been trying these things during the last weeks. I think you can try to set up a friction trigger to limit the maximal deceleration (https://carla.readthedocs.io/en/latest/tuto_G_add_friction_triggers/). Or maybe you can try to adjust the parameters in class "carla.WheelPhysicsControl". But if you want to control the deceleration more precisely, I have been failed to find a solution. It seems it will always lead to a full brake, no matter how you change the value of brake.

lucasnmatias, 2020-08-06:
Thank you very much. Yes, I've been trying to work with PhysicsControl parameters but it's hard to find good/smooth solution. All I have read right now leads to implementing the full PID controller which would take some time.

Glazyee, 2020-08-07:
I am not sure if a PID controller would solve this problem. For me, the dynamic model of the vehicle is also a little questionable. And the actuator "Brake" could only be set to brake or no brake, no matter you input a value like 0.7, 0.1 for the brake. Hope it will work in your situation.

germanros1987, 2020-08-19:
@lucasnmatias you would like to re-define the vehicle maximum acceleration? The parameters that we are using for our vehicles are the ones provided by OEMs. Could you elaborate a bit more and explain what is the actual problem?

munirjojoverge, 2020-09-08:
@germanros1987 , where can the vehicle maximum acceleration be found? I've looked on the VehiclePhysicsControl.

closed 2020-11-16 by stale[bot]

HaoZhouGT, 2021-03-03:
> I am not sure if a PID controller would solve this problem. For me, the dynamic model of the vehicle is also a little questionable. And the actuator "Brake" could only be set to brake or no brake, no matter you input a value like 0.7, 0.1 for the brake. Hope it will work in your situation. 

Hello sir, I found the same issue. Any brake value gives the same decelerations. Did you solve the brake issue in the end?

PatrickBayerOTH, 2022-05-23:
I have  the same problem on carla0.9.13.
Acceleration and Deceleration of a car controlled by the autopilot are very high. Can the maximum acceleration be set somehow?

MaisJamal, 2022-10-17:
Hello,
Has anyone managed to solve this issue? I also noticed that any brake value gives the same deceleration and it's unrealistic.

Glazyee, 2022-10-19:
> 

I managed to introduce a new vehicle dynamics model through co-simulation. But I just noticed that official support for PyChrono dynamic models has been added after some updates. Maybe you could check out that model and see if it could help you.

ClementLeBihan, 2023-03-29:
Same issue on my side. Is a trick to have better control on max acceleration and deceleration ?

TingzheYu, 2025-01-15:
I meet this problem too. No matter the brake is set 0.1 or 0.7, the resulted deceleration is the same. Is any official explanation about that?