carla-simulator/carla #3170: Carla crashes when running examples that use pygame

ivannson, 2020-08-06:
Carla version: 0.9.9
System: Windows 10, 4gb Nvidia GPU

I installed Carla and Python API, it seems to work well and I am able to run the `spawn_npc` and `dynamic_weather` examples. 

When I try running `manual_control` example, a pygame window is opened and UE4 either crashes straight away, or an image is displayed, then horizontal black lines appear and then UE crashes.

When running `automatic_control`, UE crashes straight away. 

I have tested this on various maps with np success. Any help would be appreciated.

germanros1987, 2020-08-19:
hi, @ivannson is the server window visible and working as expected?

If this is a problem with pygame, have you tried different versions of pygame? @marcgpuig have you seen anything like this before?

ivannson, 2020-08-20:
@germanros1987 yes, server window was visible. I think the issue might have been with the server and possible overload? The simulation runs fine if I press Launch instead of Play in UE4. 

I have not tried different versions of pygame yet.

zchrissirhcz, 2020-10-20:
pygame 2.0.0.dev14 (the latest version, with python 3.7.7)
CARLA 0.9.10
Windows 10

Still suffering crash issue with `manual_control.py` even if doing nothing, just wait about 10 seconds, my CPU reached 100%.

My machine is with:
- CPU: i7-4790
- GPU: NVIDIA GTX 970
- GPU Driver: 456.38

ivannson, 2020-11-02:
@zchrissirhcz Are you running it inside the Editor or as a separate process (i.e. in a new window)? Running it as a separate process seemed to do it for me as a workaround.

zchrissirhcz, 2020-11-02:
@ivannson I was running it as a separate process, not in the UE4 editor.

germanros1987, 2021-01-26:
@Axel1092 could you please follow up on this?

Axel1092, 2021-01-27:
@zchrissirhcz have you been able to fix the issue? the gtx 970 has only 4gb which is starting to be a bit short. Every camera you spawn in the server increases the gpu's memory required. Try running the `no_rendering_mode.py` which is similar to the `manual_control.py` but without any camera sensors.

closed 2021-07-21 by stale[bot]

dev3225, 2024-01-10:
any fix for these scripts which use pygame window? 
I get rpc server bind errors

jayd972, 2024-06-05:
I have installed Carla 0.9.15 and I am also having the same issue. Can anyone suggest some solution to this issue?