carla-simulator/carla #3171: assertion error when trying to run carla on windows

omarchakroun, 2020-08-06:
Hi there,
i'm newby in carla really newby and just get it installed on windows based on the steps on the documentation. i was trying to run town03 example to make sure that everything is working but i got an assertion error. when generating the project and compiling it under VS there is no error. can someone please help.
thank you and best regards.

[image]

germanros1987, 2020-08-19:
Sorry for the delay, we are in the middle of the vacation season. @Axel1092 could you help with this?

omarchakroun, 2020-08-19:
Hi germanros, 
Yesterday in fact, i was able to resolve the issue. I was using the wrong asset package. I was using the assets from the 0.9.9 with carla version 0.9.10. 
Thank you again for the support,
Omar

closed 2020-08-26 by Axel1092