carla-simulator/carla #3172: Weird dynamic performance of the vehicle

Glazyee, 2020-08-07:
Hi, as you can see in the diagram, VW and Kia are two vehicles that I have spawned in the scenario. For example, the value of the throttle of VW is almost set to 1 in this case, but the acceleration VW varies unreasonably (dark blue line) during this process. Could someone kindly explain to me, how come this happens? Thank you so much.
[image]

BigTailFox, 2020-08-12:
Hello, 
I have also met this problem, I guess the relationship between `throttle` and `acceleration` of a Carla vehicle is nonlinear, and the auto gear-box may also have effection. I am now using a PID controller to follow the expected acceleration. Do you get accelerations from an IMU sensor or the Actor API, they don't match either.

Glazyee, 2020-08-12:
Hi, thanks a lot for your information. I got the accelerations from Actor API and also set the auto gearbox as True. The velocity of the vehicle gets down from 140km/h to about 70km/h at full throttle, then start to accelerate a little bit. For me, it's still strange.
So, by using a PID controller, is the result acceptable for you now?

BigTailFox, 2020-08-12:
A bit weird for the negative acc with the positive throttle in this graph. I haven't met this situation. Turning off auto gear-box and setting a fixed gear ratio, checking the brake/handbrake and the road gradient/friction might help.

germanros1987, 2020-08-28:
Sorry for joining the party late. We are catching up after the summer break. @jackbart94 and your help is required to clarify this doubt.

jackbart94, 2020-09-03:
Hi,

Sorry for taking some time, I was wondering what script are you using for this particular example? Is Traffic Manager involved, or are you using one of the agents? It's very important because the weird behavior of the throttle might be because the PID isn't tuned properly, and might be one of the older agents.

Glazyee, 2020-10-23:
> A bit weird for the negative acc with the positive throttle in this graph. I haven't met this situation. Turning off auto gear-box and setting a fixed gear ratio, checking the brake/handbrake and the road gradient/friction might help.

Thanks a lot for the advice. And sorry for the late feedback. I've been stuck in other works in the last two months. It seems a little better but still problematic by adjusting the gearbox setting. As for the other parameter, they should not be causality.

Glazyee, 2020-10-23:
> r

Glazyee, 2020-10-23:
> Hi,
>
> Sorry for taking some time, I was wondering what script are you using for this particular example? Is Traffic Manager involved, or are you using one of the agents? It's very important because [...]

Hi,

sorry for the la.. late reply. I use also a package called scenario runner. And it offers a function to define behaviors of vehicle, such as waypointfollower, keepvelocity and so on. And in this context, a PID-controller for the velocity was applied.
I suppose that the problems should not come from the behavior of the throttle or the PID. Because even if the throttle was set to 100%, it came to a deceleration too.

closed 2021-04-27 by stale[bot]