carla-simulator/carla #3173: error: XDG_RUNTIME_DIR not set in the environment. in ssh terminal ubuntu 16.04

nuomizai, 2020-08-09:
I tried to run CarlaUE4.sh in a ubuntu16.04 server through ssh terminal, but got

4.24.3-0+++UE4+Release-4.24 518 0
Disabling core dumps.
error: XDG_RUNTIME_DIR not set in the environment.

I also tried the command `DISPLAY= ./CarlaUE4.sh`, but the error still occurred and carla can not be run. Does anyone know the reason and how to fix it? I've tried to install the ubuntu-desktop through ssh terminal.

germanros1987, 2020-08-28:
Sorry for the late response. Which version of CARLA is this? Is the environment variable SDL_VIDEODRIVER=offscreen ?
Could you try forcing the server to use OpenGL? i.e. ./CarlaUE4.sh -opengl

nuomizai, 2020-08-29:
@germanros1987 Thank you for your reply. I use the version carla:0.9.9.3, and I've tried other carla version, such as carla:0.9.7 or carla:0.9.9.1, but the error still occured. I printed the environment variable SDL_VIDEODRIVER by the command

`echo $SDL_VIDEODRIVER `

but an empty string was returned.

I also tried the command 

`./CarlaUE4.sh -opengl` This time, this XDG_RUNTIME_DIR error was gone, but it seemed that carla server exited and nothing error information was returned:

```
luban@k8s-deploy-emygwl-1593499612785-dd99fd8db-6lbbd:~/nfs/volume-363-3/zhaoyinuo/carla_0.9.9.3$ ./CarlaUE4.sh -opengl
4.24.3-0+++UE4+Release-4.24 518 0
Disabling core dumps.
luban@k8s-deploy-emygwl-1593499612785-dd99fd8db-6lbbd:~/nfs/volume-363-3/zhaoyinuo/carla_0.9.9.3$

```

closed 2021-04-27 by stale[bot]

bonniesjli, 2021-05-01:
Any update on this?
I'm having the same issues running CARLA_0.9.6 on virtual machines. -opengl still give the same error

$ DISPLAY= ./CarlaUE4.sh -opengl
4.22.1-0+++UE4+Release-4.22 517 0
Disabling core dumps.
error: XDG_RUNTIME_DIR not set in the environment.
Signal 11 caught.
Malloc Size=65538 LargeMemoryPoolOffset=65554
CommonUnixCrashHandler: Signal=11
Malloc Size=65535 LargeMemoryPoolOffset=131119
Malloc Size=53392 LargeMemoryPoolOffset=184528
Engine crash handling finished; re-raising signal 11 for the default handler. Good bye.

schissmantics, 2021-08-28:
I'm having the same error on google colab, [there](https://colab.research.google.com/drive/1fgDCgEhuuwv1HA4OYCWi9BgHWPuu8Riq?usp=sharing)

roshambo919, 2022-06-07:
I was receiving this error on ubuntu 20.04 using carla 0.9.12 from docker, but seems like 0.9.13 is working for me so far.