carla-simulator/carla #3174: How to Load LAS data in carla simulator

Bhavani56, 2020-08-10:


Bhavani56, 2020-08-10:
Hi All, i want to load LAS data in carla simulator for this am using config.py script i tried to modify the script.

elif args.las_path is not None:
        if os.path.exists(args.las_path):
            with open(args.las_path,encoding='ANSI') as od_file:
                try:
                    data = od_file.read()
                except OSError:
                    print('file could not be readed.')
                    sys.exit()
            print('load opendrive map %r.' % os.path.basename(args.las_path))
            vertex_distance = 2.0  # in meters
            max_road_length = 50.0 # in meters
            wall_height = 1.0      # in meters
            extra_width = 0.6      # in meters
            world = client.generate_opendrive_world(
                data, carla.OpendriveGenerationParameters(
                    vertex_distance=vertex_distance,
                    max_road_length=max_road_length,
                    wall_height=wall_height,
                    additional_width=extra_width,
                    smooth_junctions=True,
                    enable_mesh_visibility=True))

this parameters i am using but getting error while loading.Thanks for any help.

ERROR: time-out of 10000ms while waiting for the simulator,make sure simulator is ready and connected to localhost:2000

germanros1987, 2020-08-28:
@Bhavani56 I will ask @Axel1092 to take a look at this, but in the meantime let me ask you an entry-level question: is the server running? If so, did you try increasing the timeout of the client to 20 sec?

Axel1092, 2020-09-04:
Hi @Bhavani56, did the increase of the timeout solve your problem? If the map is very large you may need to increase the timeout even further.

closed 2020-11-14 by stale[bot]