carla-simulator/carla #3176: Throws error when try to convert osm to odr using carla.Osm2Odr.convert(osm_data, settings) it says no such attribute in carla

sp062, 2020-08-10:
Hi 

Below is the code for converting OSM to ODR also imported carla . But it gives error No such Attributes in Carla.

# Read the .osm data
f = open("C:/map.osm","r",encoding='ANSI')
osm_data = f.read()
f.close()

# Define the desired settings. In this case, default values.
settings = carla.Osm2OdrSettings()
# Convert to .xodr
xodr_data = carla.Osm2Odr.convert(osm_data, settings)

# save opendrive file
f = open("C:/test1.txt","w")
f.write(xodr_data)
f.close()

nevrax, 2020-08-16:
same error. i'm using windows binary CARLA_0.9.9.4.zip from [CARLA 0.9.9 (development)](https://github.com/carla-simulator/carla/releases/tag/0.9.9) release
```
AttributeError                            Traceback (most recent call last)
<ipython-input-17-ee30e86a7fc0> in <module>
      7 
      8 # Define the desired settings. In this case, default values.
----> 9 settings = carla.Osm2OdrSettings()
     10 # Convert to .xodr
     11 xodr_data = carla.Osm2Odr.convert(osm_data, settings)

AttributeError: module 'carla' has no attribute 'Osm2OdrSettings'
```

revikiran1991, 2020-08-19:
Same error:
Which version of carla will support OSM? I tried it in 0.9.9.4 (built on ubuntu 18.04 ). 
`Traceback (most recent call last):
  File "ConvOSM2OpnDrive.py", line 21, in <module>
    settings = carla.Osm2OdrSettings()
AttributeError: module 'carla' has no attribute 'Osm2OdrSettings' `

XieJinLiang, 2020-08-26:
This feature is only available in the unreleased 0.9.10-pre version：[0.9.10-pre](https://github.com/carla-simulator/carla/blob/master/Docs/download.md#pre-release-0910)

germanros1987, 2020-08-28:
@XieJinLiang is correct. We have not made the official announcement yet because it is in a pre-release state, but feel free to play with the feature and give us feedback!

closed 2020-08-28 by germanros1987