carla-simulator/carla #3177: Connection Error in Building

Gatsby23, 2020-08-11:
I'm new begininger in Carla, and follow the building process, I get this error, how to fix it ?
[image]

thytless, 2020-08-11:
Hey would you please provide detailed info about your problem such as in which step did this happen, and how could we reproduce it?
Looks like you are running into a network trouble though...

germanros1987, 2020-08-28:
@Gatsby23 is this issue still bothering you or should we just close it?

Gatsby23, 2020-08-29:
> @Gatsby23 is this issue still bothering you or should we just close it?

OK, I can build this it now, I close the issue.

closed 2020-08-29 by Gatsby23