carla-simulator/carla #3180: carla simulator 0.9.9 shows fatal error at start

patdee, 2020-08-12:
I have updated edge on windows 10 and carla simulator crashed with fatal error.  I have installed pre compiled version downloaded from carla site.

[UE4CC-Windows-B680E33C415A31F10E902C848AC7AF08_0000.zip](https://github.com/carla-simulator/carla/files/5061194/UE4CC-Windows-B680E33C415A31F10E902C848AC7AF08_0000.zip)

Windows build details
windows 10 version 2004 build 19041.423

germanros1987, 2020-08-28:
Hi @patdee, sorry that you are having problems. Which GPU are you using?

Kin-Zhang, 2020-12-11:
I have the same question too!
The Fatal error just show after I click .exe file

```
Unhandled Exception: 0xe06d7363

0x00007ffe106c3b29 KERNELBASE.dll!UnknownFunction []
0x00007ffe030661e0 VCRUNTIME140.dll!UnknownFunction []
0x00007ff6069c2802 CarlaUE4-Win64-Shipping.exe!clmdep_asio::detail::throw_exception<std::system_error>() []
0x00007ff6069c6070 CarlaUE4-Win64-Shipping.exe!clmdep_asio::detail::do_throw_error() []
0x00007ff6069c2bf7 CarlaUE4-Win64-Shipping.exe!clmdep_asio::basic_socket_acceptor<clmdep_asio::ip::tcp,clmdep_asio::socket_acceptor_service<clmdep_asio::ip::tcp> >::basic_socket_acceptor<clmdep_asio::ip::tcp,clmdep_asio::socket_acceptor_service<clmdep_asio::ip::tcp> >() []
0x00007ff6069c30c6 CarlaUE4-Win64-Shipping.exe!rpc::server::server() []
0x00007ff604d64c74 CarlaUE4-Win64-Shipping.exe!FCarlaServer::FPimpl::FPimpl() [c:\jenkins\workspace\carla_0.9.10\unreal\carlaue4\plugins\carla\source\carla\server\carlaserver.cpp:70]
0x00007ff604dcccf4 CarlaUE4-Win64-Shipping.exe!FCarlaEngine::NotifyInitGame() [c:\jenkins\workspace\carla_0.9.10\unreal\carlaue4\plugins\carla\source\carla\game\carlaengine.cpp:60]
0x00007ff604dba2ae CarlaUE4-Win64-Shipping.exe!ACarlaGameModeBase::InitGame() [c:\jenkins\workspace\carla_0.9.10\unreal\carlaue4\plugins\carla\source\carla\game\carlagamemodebase.cpp:100]
0x00007ff6066575fb CarlaUE4-Win64-Shipping.exe!UWorld::InitializeActorsForPlay() []
0x00007ff60660ff8a CarlaUE4-Win64-Shipping.exe!UEngine::LoadMap() []
0x00007ff606602273 CarlaUE4-Win64-Shipping.exe!UEngine::Browse() []
0x00007ff606234258 CarlaUE4-Win64-Shipping.exe!UGameInstance::StartGameInstance() []
0x00007ff604cfbb30 CarlaUE4-Win64-Shipping.exe!FEngineLoop::Init() []
0x00007ff604d027b7 CarlaUE4-Win64-Shipping.exe!GuardedMain() []
0x00007ff604d028da CarlaUE4-Win64-Shipping.exe!GuardedMainWrapper() []
0x00007ff604d0c26b CarlaUE4-Win64-Shipping.exe!WinMain() []
0x00007ff606f07e5e CarlaUE4-Win64-Shipping.exe!__scrt_common_main_seh() [d:\agent\_work\2\s\src\vctools\crt\vcstartup\src\startup\exe_common.inl:288]
0x00007ffe11c57c24 KERNEL32.DLL!UnknownFunction []
0x00007ffe136ed4d1 ntdll.dll!UnknownFunction []
```
GPU: GTX 1060Ti
CPU: i7-9750H
RAM: 16GB
System: Window 10
[image]

bernatx, 2020-12-11:
Hi, it seems that the RPC server can not be started. Can you check your port for RPC server (by default 2000) is open and free?

Kin-Zhang, 2020-12-13:
> Hi, it seems that the RPC server can not be started. Can you check your port for RPC server (by default 2000) is open and free?

Sorry, since I'm a freshman on RPC, I searched to check this way 
RPC Service: [cmd->services.msc ->Rpc & Rpc Locator & TCP/IP NetBIOS Helper all open]
Port: [cmd -> netstat -na -> I didn't find anything wrong, port 2000 is free since it didn't show up at here]
Please tell me if I did the wrong check way

ptr-br, 2021-01-19:
I have the same issue as Kin-Zhang (CARLA 0.9.11).

How did you even manage to get the error output?
After I hit the .exe only the fatal error is displayed and I'm not able to find any logs ...

bernatx, 2021-01-19:
The RPC server is launched when the CARLA server runs.
Just in case you have a problem with port 2000 (because it is in use), you can launch the CARLA server to use another port for RPC server with this parameter:

`CarlaUE4.exe -carla-rpc-port=3000`

Use any free port you have and see what happens.

Regards
Bernat

ptr-br, 2021-01-20:
Modifying the port doesn't change anything for me...
The portable version of CARLA 0.9.9.4 is running fine without an error while starting.

Thanks for your help.

Regards
Peter

Kait0, 2021-04-30:
Hi,

I would like to report a similar issue:
When I run CARLA for longer periods of time (>1hour, also sometimes happend earlier when I was debugging), with a leaderboard client, the CARLA server crashes with the same "Fatal Error!" message as shown above.
I am using CARLA version 0.9.10.1
System: Windows 10
CPU: Intel i7 6th gen and 9th gen (happens on both machines)
GPU: GTX 980, RTX 2070 (Laptop)
RAM: 16 GB

I don't know much about RPC but since CARLA is running for a while it seems unlikely that the issue is that the port is becoming unavailable.

Best Regards
Bernhard

PS: Is there a way to catch these crashes from the client side (so I can automatically restart the server when it happens)?

Kin-Zhang, 2021-05-12:
> The RPC server is launched when the CARLA server runs.
> Just in case you have a problem with port 2000 (because it is in use), you can launch the CARLA server to use another port for RPC server with [...]

It works!!! Thanks! But I still what to know how to check or who is using port 2000 since like I mentioned before about:
Port: [cmd -> netstat -na -> I didn't find anything wrong, port 2000 is free since it didn't show up at here]
Thanks again! bernatx

Kin-Zhang, 2021-05-12:
> I have the same issue as Kin-Zhang (CARLA 0.9.11).
> 
> How did you even manage to get the error output?
> After I hit the .exe only the fatal error is displayed and I'm not able to find any logs ...

When you click Confirm/"確認" it will automatically copy something in the clipboard, so after you click 確認 you can open note and CTRL+V and you will find the log

Kait0, 2021-05-13:
Good to know. 
One of the errors I encountered:
 Does not seem to be the same as the one you showed above. Perhaps a different error leading to the same symptom

LowLevelFatalError [File:Unknown] [Line: 853] 
Rendering thread exception:
Fatal error!

Unhandled Exception: EXCEPTION_ACCESS_VIOLATION reading address 0xe98e8508

0x00007ff6455cdb8e CarlaUE4-Win64-Shipping.exe!FInstancedStaticMeshSceneProxy::~FInstancedStaticMeshSceneProxy() []
0x00007ff6455cf6a4 CarlaUE4-Win64-Shipping.exe!FInstancedStaticMeshSceneProxy::`scalar deleting destructor'() []
0x00007ff644c3ce41 CarlaUE4-Win64-Shipping.exe!FScene::UpdateAllPrimitiveSceneInfos() []
0x00007ff644c2505b CarlaUE4-Win64-Shipping.exe!TRDGLambdaPass<FGenerateSkyLightVisibilityRaysCS::FParameters,<lambda_b2336f278bcd1ed1ee1f0d3535b2ffb0> >::ExecuteImpl() []
0x00007ff6443cfc99 CarlaUE4-Win64-Shipping.exe!FNamedTaskThread::ProcessTasksNamedThread() []
0x00007ff6443cfda3 CarlaUE4-Win64-Shipping.exe!FNamedTaskThread::ProcessTasksUntilQuit() []
0x00007ff644fbd282 CarlaUE4-Win64-Shipping.exe!RenderingThreadMain() []
0x00007ff644fbde58 CarlaUE4-Win64-Shipping.exe!FRenderingThread::Run() []
0x00007ff6445018ab CarlaUE4-Win64-Shipping.exe!FRunnableThreadWin::Run() []
0x00007ff644501684 CarlaUE4-Win64-Shipping.exe!FRunnableThreadWin::GuardedRun() []
0x00007ffe04a17034 KERNEL32.DLL!UnknownFunction []
0x00007ffe057c2651 ntdll.dll!UnknownFunction []

Kin-Zhang, 2021-08-04:
> Sorry, since I'm a freshman on RPC, I searched to check this way
> RPC Service: [cmd->services.msc ->Rpc & Rpc Locator & TCP/IP NetBIOS Helper all open]
> Port: [cmd -> netstat -na -> I didn't find [...]

OK Now, I know how to check the TPC Port for RPC on Windows, open the cmd:
`netstat -ano`

[image]

And follow the PID thread I found the sqlsevr.exe occupied Port 2000 which is windows server

mossr, 2022-02-27:
@bernatx I occasionally get the same exact error as @Kait0 when calling `client.load_world("Town04")` as an example (on Windows 10, running `CarlaUE4.exe`). Changing the RPC port does not fix it.

Note, there is no obvious steps to recreate this error. It happens about every 1/5th time I open CarlaUE4.exe and run the `load_world` code in Python, and the other 4/5 times it works without issue 🤷‍♂️

@bernatx any insights or things to try and debug this further on my end?

```
LowLevelFatalError [File:Unknown] [Line: 853] 
Rendering thread exception:
Fatal error!

Unhandled Exception: EXCEPTION_ACCESS_VIOLATION reading address 0xdb6c3f08

0x00007ff7d2b7413e CarlaUE4-Win64-Shipping.exe!FInstancedStaticMeshSceneProxy::~FInstancedStaticMeshSceneProxy() [c:\program files\epic games\ue_4.24\engine\source\runtime\engine\private\instancedstaticmesh.h:527]
0x00007ff7d2b75c1a CarlaUE4-Win64-Shipping.exe!FHierarchicalStaticMeshSceneProxy::`scalar deleting destructor'() []
0x00007ff7d21e7071 CarlaUE4-Win64-Shipping.exe!FScene::UpdateAllPrimitiveSceneInfos() [c:\program files\epic games\ue_4.24\engine\source\runtime\renderer\private\rendererscene.cpp:3949]
0x00007ff7d21cf28b CarlaUE4-Win64-Shipping.exe!TGraphTask<TEnqueueUniqueRenderCommandType<`FScene::Release'::`2'::FReleaseCommandName,<lambda_4674698005fc94d565566ee71a6e420d> > >::ExecuteTask() [c:\program files\epic games\ue_4.24\engine\source\runtime\core\public\async\taskgraphinterfaces.h:847]
0x00007ff7d197c5e9 CarlaUE4-Win64-Shipping.exe!FNamedTaskThread::ProcessTasksNamedThread() [c:\program files\epic games\ue_4.24\engine\source\runtime\core\private\async\taskgraph.cpp:686]
0x00007ff7d197c6f3 CarlaUE4-Win64-Shipping.exe!FNamedTaskThread::ProcessTasksUntilQuit() [c:\program files\epic games\ue_4.24\engine\source\runtime\core\private\async\taskgraph.cpp:583]
0x00007ff7d2566122 CarlaUE4-Win64-Shipping.exe!RenderingThreadMain() [c:\program files\epic games\ue_4.24\engine\source\runtime\rendercore\private\renderingthread.cpp:340]
0x00007ff7d2566cf8 CarlaUE4-Win64-Shipping.exe!FRenderingThread::Run() [c:\program files\epic games\ue_4.24\engine\source\runtime\rendercore\private\renderingthread.cpp:471]
0x00007ff7d1aad3db CarlaUE4-Win64-Shipping.exe!FRunnableThreadWin::Run() [c:\program files\epic games\ue_4.24\engine\source\runtime\core\private\windows\windowsrunnablethread.cpp:96]
0x00007ff7d1aad1b4 CarlaUE4-Win64-Shipping.exe!FRunnableThreadWin::GuardedRun() [c:\program files\epic games\ue_4.24\engine\source\runtime\core\private\windows\windowsrunnablethread.cpp:53]
0x00007ff886c07034 KERNEL32.DLL!UnknownFunction []
0x00007ff887e22651 ntdll.dll!UnknownFunction []
```

Thank you.

abdooo786, 2022-04-22:
i don't know why , but it worked with me using "./CarlaUE4.sh -opengl13"

ABD-01, 2024-03-31:
Hi
I too am facing a similar issue
Logs:
```
Fatal error!

Unhandled Exception: EXCEPTION_ACCESS_VIOLATION reading address 0x00000212d8a15420

0x00007ffc37a31490 VCRUNTIME140.dll!UnknownFunction []
0x00007ff7d6f66dfe CarlaUE4-Win64-Shipping.exe!<lambda_5ce23935cadd5808955cd519361e8be7>::operator()() [C:\jenkins\workspace\carla_0.9.15\Unreal\CarlaUE4\Plugins\Carla\Source\Carla\Sensor\PixelReader.h:157]
0x00007ff7d6f687c9 CarlaUE4-Win64-Shipping.exe!<lambda_61c31b803b4dc0f32cb83b8fed0089c7>::operator()() [C:\jenkins\workspace\carla_0.9.15\Unreal\CarlaUE4\Plugins\Carla\Source\Carla\Sensor\PixelReader.cpp:77]
0x00007ff7d7714817 CarlaUE4-Win64-Shipping.exe!TGraphTask<FAsyncGraphTask>::ExecuteTask() [C:\UE_4.26\Engine\Source\Runtime\Core\Public\Async\TaskGraphInterfaces.h:887]
0x00007ff7d7721429 CarlaUE4-Win64-Shipping.exe!FNamedTaskThread::ProcessTasksNamedThread() [C:\UE_4.26\Engine\Source\Runtime\Core\Private\Async\TaskGraph.cpp:710]
0x00007ff7d77216be CarlaUE4-Win64-Shipping.exe!FNamedTaskThread::ProcessTasksUntilQuit() [C:\UE_4.26\Engine\Source\Runtime\Core\Private\Async\TaskGraph.cpp:601]
0x00007ff7d787b8dd CarlaUE4-Win64-Shipping.exe!FStatsThread::Run() [C:\UE_4.26\Engine\Source\Runtime\Core\Private\Stats\Stats2.cpp:906]
0x00007ff7d78c05fb CarlaUE4-Win64-Shipping.exe!FRunnableThreadWin::Run() [C:\UE_4.26\Engine\Source\Runtime\Core\Private\Windows\WindowsRunnableThread.cpp:86]
0x00007ff7d78bbc41 CarlaUE4-Win64-Shipping.exe!FRunnableThreadWin::GuardedRun() [C:\UE_4.26\Engine\Source\Runtime\Core\Private\Windows\WindowsRunnableThread.cpp:35]
0x00007ffc4c2e257d KERNEL32.DLL!UnknownFunction []
0x00007ffc4deeaa58 ntdll.dll!UnknownFunction []

Crash in runnable thread StatsThread
```

Before running Carla I cross-checked no process is using port `2000` using `netstat -ano`
The town was loaded, then crash occurred.
 Changing the port also results in the same consequences.

MarcelPiNacy-CVC, 2024-05-10:
> Hi I too am facing a similar issue Logs:
>
> ```
> Fatal error!
>
> Unhandled Exception: EXCEPTION_ACCESS_VIOLATION reading address 0x00000212d8a15420
>
> 0x00007ffc37a31490 VCRUNTIME140.dll!UnknownFunction [] [...]

Hi,

this is actually not the same kind of problem, please open another issue for this.
I'm closing this one for now, feel free to reopen it.

Marcel

closed 2024-05-10 by MarcelPiNacy-CVC