carla-simulator/carla #3182: How visualize the carla road network map on rviz?

XieJinLiang, 2020-08-12:
Hi,
Is there a way to visualize the carla opendrive map on rviz?What  message type should I use?Or can I only draw it line by line manually?

germanros1987, 2020-08-29:
Hi @XieJinLiang there is no specific message for that, but you could take a look at this: https://github.com/carla-simulator/carlaviz

The message should be compatible with rviz, so that is your best starting point.

closed 2020-08-29 by germanros1987

Roosstef, 2020-09-30:
@XieJinLiang Did you find a way to visualize it on rviz?

minhaj6, 2025-04-11:
@XieJinLiang @Roosstef did you guys figure out how to draw that on rviz? 

I need to visualize the vehicle trajectories. I was thinking about the Carla_Painter in CarlaViz, but unfortunately the project looks abandoned and the author hasn't implemented the functionality.