carla-simulator/carla #3183: Collision Sensor Accuracy!

KasraMokhtari2112, 2020-08-13:
I have one question about the collision sensor. I am interested to catch the collision between the ego vehicle and the pedestrians. However, this sensor is not accurate in most cases. Does Anyone know why it’s working like this? And any ideas on how to improve the accuracy of the collision sensor? I am in desperate need of help at this moment. Thanks for your time.

thytless, 2020-08-13:
As far as I know the collision sensor calculates whether there is overlap between bounding boxes, so its impl decides that it cannot be precise.

KasraMokhtari2112, 2020-08-13:
@thytless I actually tried to change the size of the bounding box (set them to zero) in Navigation.cpp and then built it from the source. However, when I print out the size of the bounding boxes of the vehicles in my python file, it still remains the same. Do you know what exactly I need to change? Thanks!

germanros1987, 2020-08-29:
Hi,

Sorry for joining the party late, but I bring some news.

Indeed, as @thytless mentioned, there are some discrepancies between the actual mesh of the object, its collision box(es) and its bounding box. The first thing to notice is that the bounding-box and the collision box are not the same (surprise!).

In the past, we were opting for having a match between the collision box and the bounding box of a vehicle. However, this seems to be a bad idea and we are moving to a new representation based on the following rules:

* Collision boxes will be tighter and the will better represent the silhouette of a vehicle. This is done to reduce false collisions.  * The bounding box could potentially be bigger than the collision box/collision template. This bounding box is mainly for groundtruth purposes and does not interfere in the physics.

We are upgrading our models to follow these rules, so some of these problems will be solved soon.

KasraMokhtari2112, 2020-08-30:
> Hi,
>
> Sorry for joining the party late, but I bring some news.
>
> Indeed, as @thytless mentioned, there are some discrepancies between the actual mesh of the object, its collision box(es) and its [...]

Thank you so much for your response. That's great news that you guys are working on this issue. In the meantime, do you know where in source codes, I can make the bounding boxes smaller?

germanros1987, 2021-01-26:
This should be now solved in 0.9.11.

closed 2021-01-26 by germanros1987

AOOOOOA, 2024-03-22:
> This should be now solved in 0.9.11.

Hi, we have met the false negative cases of the collision sensor in Carla0.9.13. The collision happened but the collision sensor failed to detect it.
It seems like the issues haven't been totally fixed or the last fix brings new problems.
I describe the issue in detail at https://github.com/carla-simulator/carla/issues/7025#issue-2052396755 . Please check.