carla-simulator/carla #3184: Building Carla in Ubuntu 18.04

KasraMokhtari2112, 2020-08-13:
Hi. I want to build Carla from the source. I followed all the steps on this link on Unreal Engine 4.24.3:
https://carla.readthedocs.io/en/latest/build_linux/ 
Everything looked fine and then I generated CarlaUE4.sh. when I am running my RL code with this generated file it seems like there is a jump (discontinuity) in the frames and sometimes the connection between my python code and the simulation will get lost. 
Any idea would be appreciated at this point since I am so confused about why this is happening!!

germanros1987, 2020-08-29:
Hi @KasraMokhtari2112, sorry for the delay. For starters, are you building your RL code using the synchronous mode?
Did you run some of our examples first? For instance, manual_control? Is everything else working? 

Please provide some further details.

KasraMokhtari2112, 2020-08-30:
> Hi @KasraMokhtari2112, sorry for the delay. For starters, are you building your RL code using the synchronous mode?
> Did you run some of our examples first? For instance, manual_control? Is everything else working?
> 
> Please provide some further details.

Thanks for your reply. I solved that issue by running Carla with fixed fps. However, the problem is after like 100-200 episodes (around 2hours), the codes crashed with a segmentation error.  I have a function that adds and destroys pedestrians after a specific time lets say 10 seconds in the environment and the code mostly crashed there. I do not use synchronous mode, do you think using that will help the stability in general?

closed 2021-04-27 by stale[bot]