carla-simulator/carla #3187: Not called on the RSS sensor callback

33v33, 2020-08-13:
Hello Carla team,

I came across an issue while trying to use the RSS sensor, while attaching it to a vehicle and I have another car nearby, I am not getting called on the sensor callback, I am on version 0.9.9.4 both server and client, and my code looks something like this:
```      
    rss_blueprint = self._kinesim.world.get_blueprint_library().find("sensor.other.rss")
    transform = carla.Transform(carla.Location(x=self.x, y=self.y, z=self.z))
    self.sensor = self._kinesim.world.spawn_actor(rss_blueprint, transform, attach_to=vehicle._vehicle)
    self.sensor.listen(self._on_rss_responses)
```
and the callback:
```
    def _on_rss_responses(self, response):
        print(response.timestamp)
        print(response.response_valid)
        print(response.proper_response)
        print(response.acceleration_restriction)
        print(response.ego_dynamics_on_route)
```

Is there something I am missing?

Thanks

germanros1987, 2020-08-29:
@fpasch could you please have a look at this?

lucasnmatias, 2020-10-19:
I'm having the same problem, when attaching RSS sensor no error is thrown but the callback function is never called.

fpasch, 2020-10-20:
Have you checked if the example script manual_control_rss.py is working as expected?

closed 2021-04-27 by stale[bot]