carla-simulator/carla #3188: Segfault on 0.9.10pre with automatic_control.py

dsche, 2020-08-13:
System: Ubuntu 20.04, NVIDIA Quadro P2000
Carla: 0.9.10pre built for Ubuntu 18.04
Command line: `gdb -ex run --args CarlaUE4/Binaries/Linux/CarlaUE4-Linux-Shipping Carla -opengl`

On connecting an example-based client (eg `automatic_control.py`), Carla segfaults.
```
Thread 1 "CarlaUE4-Linux-" received signal SIGSEGV, Segmentation fault.
__memmove_avx_unaligned_erms () at ../sysdeps/x86_64/multiarch/memmove-vec-unaligned-erms.S:262
262     ../sysdeps/x86_64/multiarch/memmove-vec-unaligned-erms.S: Datei oder Verzeichnis nicht gefunden.
(gdb) bt
#0  __memmove_avx_unaligned_erms () at ../sysdeps/x86_64/multiarch/memmove-vec-unaligned-erms.S:262
#1  0x0000000001eb7ede in carla::Buffer::copy_from (this=<optimized out>, offset=12, data=0x0, size=5120)
    at /home/axel/Documents/jenkins/Unreal/CarlaUE4/Plugins/Carla/CarlaDependencies/include/carla/Buffer.h:351
#2  ASceneCaptureSensor::CopyTextureFromAtlas (this=0x7fff57973f80, Buffer=..., AtlasImage=..., AtlasTextureWidth=<optimized out>)
    at /home/axel/Documents/jenkins/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Sensor/SceneCaptureSensor.cpp:586
#3  0x0000000001eb7375 in ASceneCaptureSensor::SendPixelsInStream<ASceneCaptureCamera> (this=0x7fff57973f80, Sensor=..., AtlasImage=empty, AtlasTextureWidth=1280)
    at /home/axel/Documents/jenkins/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Sensor/SceneCaptureSensor.h:285
#4  0x0000000001e92b3e in ACarlaGameModeBase::SendAtlas (this=<optimized out>) at /home/axel/Documents/jenkins/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Game/CarlaGameModeBase.cpp:475
#5  ACarlaGameModeBase::Tick (this=0x7fff56a9edc0, DeltaSeconds=0.106841892) at /home/axel/Documents/jenkins/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Game/CarlaGameModeBase.cpp:222
#6  0x0000000003c33a5a in FTickFunctionTask::DoTask (this=0x7fff55779c60, CurrentThread=ENamedThreads::GameThread, MyCompletionGraphEvent=...)
    at Runtime/Engine/Private/TickTaskManager.cpp:285
#7  TGraphTask<FTickFunctionTask>::ExecuteTask (this=0x7fff55779c50, NewTasks=empty, CurrentThread=ENamedThreads::GameThread) at Runtime/Core/Public/Async/TaskGraphInterfaces.h:847
#8  0x00000000021a7afb in FBaseGraphTask::Execute (NewTasks=..., CurrentThread=5120, this=<optimized out>) at Runtime/Core/Public/Async/TaskGraphInterfaces.h:514
#9  FNamedTaskThread::ProcessTasksNamedThread (this=0x7ffff5a684d0, QueueIndex=0, bAllowStall=<optimized out>) at Runtime/Core/Private/Async/TaskGraph.cpp:686
#10 0x00000000021a7763 in FNamedTaskThread::ProcessTasksUntilQuit (this=0x7ffff5a684d0, QueueIndex=0) at Runtime/Core/Private/Async/TaskGraph.cpp:582
#11 0x00000000021a6ff7 in FTaskGraphImplementation::ProcessThreadUntilRequestReturn (this=<optimized out>, CurrentThread=<optimized out>) at Runtime/Core/Private/Async/TaskGraph.cpp:1414
#12 FTaskGraphImplementation::WaitUntilTasksComplete (this=0x7ffff5b1e020, Tasks=..., CurrentThreadIfKnown=<optimized out>) at Runtime/Core/Private/Async/TaskGraph.cpp:1463
#13 0x0000000003c3025c in FTickTaskSequencer::ReleaseTickGroup (this=0x5249b58 <FTickTaskSequencer::Get()::SingletonInstance>, WorldTickGroup=TG_PrePhysics, 
    bBlockTillComplete=<optimized out>) at Runtime/Engine/Private/TickTaskManager.cpp:559
#14 0x0000000003c2dc35 in FTickTaskManager::RunTickGroup (this=0x5249b00 <FTickTaskManager::Get()::SingletonInstance>, Group=TG_PrePhysics, bBlockTillComplete=<optimized out>)
    at Runtime/Engine/Private/TickTaskManager.cpp:1527
#15 0x0000000003866f8b in UWorld::RunTickGroup (this=<optimized out>, Group=TG_PrePhysics, bBlockTillComplete=<optimized out>) at Runtime/Engine/Private/LevelTick.cpp:781
#16 UWorld::Tick (this=<optimized out>, TickType=LEVELTICK_All, DeltaSeconds=0.106841892) at Runtime/Engine/Private/LevelTick.cpp:1552
#17 0x0000000003773b19 in UGameEngine::Tick (this=0x7fffe7450100, DeltaSeconds=<optimized out>, bIdleMode=<optimized out>) at Runtime/Engine/Private/GameEngine.cpp:1665
#18 0x0000000001e55e8e in FEngineLoop::Tick (this=0x50463b0 <GEngineLoop>) at Runtime/Launch/Private/LaunchEngineLoop.cpp:4479
#19 0x0000000001e5720a in EngineTick () at Runtime/Launch/Private/Launch.cpp:62
#20 GuardedMain (CmdLine=<optimized out>) at Runtime/Launch/Private/Launch.cpp:175
#21 0x0000000003f356a8 in CommonUnixMain (argc=<optimized out>, argv=0x7fffffffda28, RealMain=0x1e56fc0 <GuardedMain(char16_t const*)>, 
    AppExitCallback=0x1e60310 <LaunchUnix_FEngineLoop_AppExit()>) at Runtime/Unix/UnixCommonStartup/Private/UnixCommonStartup.cpp:264
#22 0x00007ffff76b80b3 in __libc_start_main (main=0x1e60300 <main(int, char**)>, argc=3, argv=0x7fffffffda28, init=<optimized out>, fini=<optimized out>, rtld_fini=<optimized out>, 
    stack_end=0x7fffffffda18) at ../csu/libc-start.c:308
#23 0x0000000001e54029 in _start ()
```

germanros1987, 2020-08-29:
@dsche we are aware of this issue and we are about the patch 0.9.10-pre. We are all experiencing problems with OpenGL. In the meantime, have you tried running the vulkan version?

dsche, 2020-08-31:
Yes, I tried it (same machine, same setup). The Vulkan version does not seem to suffer from the above mentioned bug, however, it is quite VRAM-hungry (rendering the Vulkan backend unusable). Running eg `spawn_npc.py` sends it into device OOM:
```
 gdb -ex run --args CarlaUE4/Binaries/Linux/CarlaUE4-Linux-Shipping Carla -vulkan
GNU gdb (Ubuntu 9.1-0ubuntu1) 9.1
Copyright (C) 2020 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
--Type <RET> for more, q to quit, c to continue without paging--c
Type "apropos word" to search for commands related to "word"...
Registered pretty printers for UE4 classes
Reading symbols from CarlaUE4/Binaries/Linux/CarlaUE4-Linux-Shipping...
Reading symbols from /home/dimitri/tools/CARLA_0_9_10-Pre_Ubuntu18/CarlaUE4/Binaries/Linux/CarlaUE4-Linux-Shipping.debug...
Starting program: /home/dimitri/tools/CARLA_0_9_10-Pre_Ubuntu18/CarlaUE4/Binaries/Linux/CarlaUE4-Linux-Shipping Carla -vulkan
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
4.24.3-0+++UE4+Release-4.24 518 0
Disabling core dumps.
[New Thread 0x7ffff6c9f700 (LWP 658266)]
[New Thread 0x7ffff2d4a700 (LWP 658267)]
[New Thread 0x7ffff2ce9700 (LWP 658268)]
[New Thread 0x7ffff2c88700 (LWP 658269)]
[New Thread 0x7ffff2c1f700 (LWP 658270)]
[New Thread 0x7ffff2baf700 (LWP 658271)]
[New Thread 0x7ffff2b3f700 (LWP 658272)]
[New Thread 0x7ffff2acf700 (LWP 658273)]
[New Thread 0x7ffff2a5f700 (LWP 658274)]
[New Thread 0x7ffff29fe700 (LWP 658275)]
[New Thread 0x7ffff299d700 (LWP 658276)]
[New Thread 0x7ffff293c700 (LWP 658277)]
[New Thread 0x7ffff28db700 (LWP 658278)]
[New Thread 0x7ffff28ba700 (LWP 658279)]
[New Thread 0x7ffff2899700 (LWP 658280)]
[New Thread 0x7ffff2878700 (LWP 658281)]
[New Thread 0x7ffff2857700 (LWP 658282)]
[New Thread 0x7ffff2836700 (LWP 658283)]
[Detaching after vfork from child process 658284]
[New Thread 0x7ffff1d84700 (LWP 658286)]
[New Thread 0x7ffff1d18700 (LWP 658287)]
[New Thread 0x7ffff1266700 (LWP 658288)]
[New Thread 0x7ffff1245700 (LWP 658289)]
[New Thread 0x7ffff1224700 (LWP 658290)]
[New Thread 0x7ffff1203700 (LWP 658291)]
[New Thread 0x7ffff11e2700 (LWP 658292)]
[New Thread 0x7fffe9ddc700 (LWP 658293)]
[New Thread 0x7fffdca7c700 (LWP 658294)]
[New Thread 0x7fffe9d5b700 (LWP 658295)]
[New Thread 0x7fffe9d3a700 (LWP 658296)]
[New Thread 0x7fffe9d19700 (LWP 658297)]
[New Thread 0x7fffe9cf8700 (LWP 658298)]
[New Thread 0x7fffe9cd7700 (LWP 658299)]
[New Thread 0x7fffe9cb6700 (LWP 658300)]
[New Thread 0x7fffe9c95700 (LWP 658301)]
[New Thread 0x7fffdb40f700 (LWP 658302)]
[New Thread 0x7fffcc96f700 (LWP 658303)]
[Thread 0x7fffcc96f700 (LWP 658303) exited]
[New Thread 0x7fffcc96f700 (LWP 658304)]
[Thread 0x7fffcc96f700 (LWP 658304) exited]
[New Thread 0x7fffcc96f700 (LWP 658305)]
[Thread 0x7fffcc96f700 (LWP 658305) exited]
[New Thread 0x7fffcc96f700 (LWP 658306)]
[... 46 lines cut ...]
#10 0x0000000002fa8c92 in FRHICommandListImmediate::CreateTexture2D (this=0x0, SizeX=<optimized out>, SizeY=<optimized out>, Format=<optimized out>, NumMips=<optimized out>, NumSamples=<optimized out>, Flags=0, CreateInfo=...) at Runtime/RHI/Public/RHICommandList.h:4120
#11 RHICreateTexture2D (SizeX=<optimized out>, SizeY=<optimized out>, Format=<optimized out>, NumMips=<optimized out>, NumSamples=<optimized out>, Flags=0, CreateInfo=...) at Runtime/RHI/Public/RHICommandList.h:5095
#12 RHICreateTargetableShaderResource2D (SizeX=856, SizeY=640, Format=<optimized out>, NumMips=1, Flags=<optimized out>, TargetableTextureFlags=<optimized out>, bForceSeparateTargetAndShaderResource=<optimized out>, bForceSharedTargetAndShaderResource=<optimized out>, CreateInfo=..., OutTargetableTexture=..., OutShaderResourceTexture=..., NumSamples=<optimized out>) at Runtime/RHI/Public/RHIUtilities.h:582
#13 0x0000000002fa0a16 in FRenderTargetPool::FindFreeElement (this=<optimized out>, RHICmdList=..., InputDesc=..., Out=..., InDebugName=<optimized out>, bDoWritableBarrier=<optimized out>, TransienceHint=<optimized out>, bDeferTextureAllocation=<optimized out>) at Runtime/RenderCore/Private/RenderTargetPool.cpp:357
#14 0x0000000002f9fe37 in FRDGBuilder::AllocateRHITextureIfNeeded (this=0x7fffdca7aff0, Texture=<optimized out>) at Runtime/RenderCore/Private/RenderGraphBuilder.cpp:510
#15 0x0000000002fa1ada in FRDGBuilder::PrepareResourcesForExecute (this=0x7fffdca7aff0, Pass=0x7fff4fbe8a70, OutRPInfo=0x7fffdca7ab78, bOutHasGraphicsOutputs=0x7fffdca7ab6f) at Runtime/RenderCore/Private/RenderGraphBuilder.cpp:965
#16 0x0000000002f9efc3 in FRDGBuilder::ExecutePass (this=0x7fffdca7aff0, Pass=0x7fff4fbe8a70) at Runtime/RenderCore/Private/RenderGraphBuilder.cpp:688
#17 0x0000000002f960cc in FRDGBuilder::Execute (this=0x7fffdca7aff0) at Runtime/RenderCore/Private/RenderGraphBuilder.cpp:185
[... cut ...]
#27 FRenderingThread::Run (this=<optimized out>) at Runtime/RenderCore/Private/RenderingThread.cpp:488
#28 0x00000000021dcbe2 in FRunnableThreadPThread::Run (this=0x7fff875a0000) at Runtime/Core/Private/HAL/PThreadRunnableThread.cpp:25
#29 0x00000000021cca9d in FRunnableThreadPThread::_ThreadProc (pThis=0x7fff875a0000) at Runtime/Core/Private/HAL/PThreadRunnableThread.h:177
#30 0x00007ffff7f74609 in start_thread (arg=<optimized out>) at pthread_create.c:477
#31 0x00007ffff77b3103 in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95
```

But I guess this is already covered by #2858.

germanros1987, 2021-01-26:
@dsche do you know if this is still happening in recent versions, such as 0.9.11?

dsche, 2021-01-27:
@germanros1987 No, I actually cannot reproduce it anymore, so I guess it is fixed now.

vishal-2000, 2021-06-25:
Seems like it was just time-out. Running it again after few seconds fixed the error in my case

PR #3211 (2020-08-28): Initial fix for OpenGL crash

#### Description

This fixes the OpenGL crash when connecting but more testing is strongly needed

Fixes #3188

#### Where has this been tested?

  * **Platform(s):** Ubuntu 18
  * **Python version(s):** 3
  * **Unreal Engine version(s):** 4.24

---
This change is [[image]](https://reviewable.io/reviews/carla-simulator/carla/3211)
closed 2021-11-17 by germanros1987