carla-simulator/carla #3191: Semantic Lidar Object ObjIdx Definition and Uniqueness

dHonerkamp, 2020-08-14:
Hi, 
I am using the semantic lidar to get instance masks. According to the documentation it returns the `index of the hit actor`. I could not find any further details. From looking through some parts of the source code this does not seem to be the actor id, but rather the `FActorInfo->Description.UId`.

Do you have some details on what exactly this id is and how unique it is? In some cases I seem to receive the same id for different instances. For both cars and pedestrians.

For an example see the two cars below which received the same id (lidar ObjIdx projected into camera perspective and overlayed with semantic camera output). I already double checked that I am using the Object Index and not the semantic tag of the semantic lidar.
[image]

Any insights would be much appreciated.

```
Client API version     = 0.9.9.4-255-gc56a7738
Simulator API version  = c56a7738 (latest docker image)
```

germanros1987, 2020-08-29:
@marcgpuig @DSantosO could you please comment on this?

DSantosO, 2020-09-01:
Hello @dHonerkamp,
This id should be unique for dynamic objects like cars and pedestrians and zero for static objects like buildings and such. If you see the same id for a car and a pedestrian, please let us know because this should not be the case.

FiodarKazhamiaka, 2020-10-20:
@DSantosO, thanks for clearing up the uniqueness. Can you comment on the definition, i.e., what it indexes into? I thought maybe its an index to the actor snapshot in `world.get_snapshot()`, but it doesn't seem to match. Can you advise on how to use the ObjIdx obtain a reference to the corresponding actor or actor snapshot?

DSantosO, 2020-10-20:
Hello @FiodarKazhamiaka,
The index was intended to the same as the index returned by the world.get_actors() but a couple of weeks ago, we discover it was not the case. We fixed in PR #3396 and you can find it working as expected already in dev branch. Sorry about that.
In branch **dsantos/lidar_bb_test**, you can find a script ([here](https://github.com/carla-simulator/carla/blob/dsantoso/lidar_bb_test/PythonAPI/examples/lidar_check_bb.py)), that compares the semantic lidar point cloud with the BB of objects. it is not in dev yet because it still needs some clean-up but maybe it can be helpful.

FiodarKazhamiaka, 2021-01-19:
Hi @DSantosO,
With Carla 0.9.11, it seems that it is no longer an object index, but rather the actor's id that being given by the semantic lidar. And if the object doesn't correspond to an actor, the id field is set to 0. This seems like a better design, and it would be great if the documentation was updated to reflect this change!

DSantosO, 2021-01-20:
Hello @FiodarKazhamiaka,

Where do you think the documentation should be updated? I have checked and in the sensor section and it already showed that the returned value is the actor id. Do you mean the id=0 when the object is not an actor?

closed 2021-07-21 by stale[bot]