carla-simulator/carla #3192: is there a semi-truck available

manonechu, 2020-08-14:
Hi, thank you for the great simulator.

I would like to know that is there a semi-truck available? Because I'd like to simulate autonomous driving for a semi-truck. If it is not available, is there a way to create or import it? Please kindly advise me.

Thank you so much for your help!

germanros1987, 2020-08-29:
Hi,

We don't have semi-trucks yet. In general, we don't have vehicles with more than 4 wheels yet. This feature is planned for future releases.

manonechu, 2020-10-26:
Hi @germanros1987, is there any good news on the development or release of semi-trucks? Thanks a lot.

XGodina, 2020-10-26:
Hi @manonechu, not yet. Now, we are working on improving our current vehicles with new materials and geometry to make them more realistic. But we have it in mind to put semi-trucks

manonechu, 2020-10-27:
Thank you for your response.

Do you have a timeline for releasing semi-trucks? Thank you! Regards

XGodina, 2020-10-27:
Hi -manonechu 

To be honest for this release I'm totally sure not. Because we want to add new vehicles of service like a taxi, ambulance, etc. But is probably when we finish this we will working on that.

closed 2020-11-11 by manonechu

nitschephil, 2021-02-16:
Hi, are there any updates regarding the implementation of more than four wheels? Our team would like to simulate a three-axle truck. Thanks!