carla-simulator/carla #3193: Runtime

Malathi15, 2020-08-15:


gsujansai, 2020-08-16:
Did you launch the carla server? You need to run CarlaUE4.sh in your Dist folder to launch it.

Malathi15, 2020-08-17:
Thanks for your reply @gsujansai 
I tried to use CARLA Simulator Linux build version as instructed in this link https://carla.readthedocs.io/en/latest/build_linux/#get-assets.
CarlaUE4.sh file not available in this CARLA Simulator folder

guruvishnuvardan, 2020-08-18:
****

germanros1987, 2020-08-29:
Hi @Malathi15,

 do you really need to build the simulator from scratch? Could you just use the most recent binary release?

smmmmi, 2020-09-21:
Same problem +1

evdoks, 2020-09-23:
The same problem on version 0.9.9.4 built from source.
I've managed to build Carla and PythonAPI on MacOS and the Carla is running, but Python examples cannot connect to the server and fail with the time-out. 

I can start Unreal Editor with `make launch` (or `make launch-only`), then launch Carla from the editor. I can see, that the UE4Editor process is listening on ports 2000 and 2001. The versions of the server and client can be read and match:

```python
import carla

client = carla.Client(host='127.0.0.1', port=2000)
server_version = client.get_server_version()
client_version = client.get_client_version()
```
I also see TCP chatter between the client and the server in Wireshark, so the server is definitely running and responding.

However, executing an example (e.g., `spawn_npc.py`) or  trying to load a city with `client.load_world('Town03')` results in a time-out:
```
    client.load_world('Town03')
RuntimeError: time-out of 10000ms while waiting for the simulator, 
make sure the simulator is ready and connected to 127.0.0.1:2000
```

agourishetty93, 2020-09-23:
Hi, Try playing the unreal engine and run the python scripts

evdoks, 2020-09-23:
Hi @agourishetty93 , this is what I am doing: I am launching Carla by clicking "Play" in UE Editor, the server seems to be starting. Is that what you mean?

Without launching Carla nothing is listening on ports 2000-2001 and `client.get_server_version()` fails with time-out.

nagasanjay, 2020-10-10:
I have the same issue, but in windows... But I couldn't find port number 2000 or 2001 in netstat though.

pardhuAj, 2020-12-07:
Hi @Malathi15, Were you able to resolve this issue. I am also facing the same issue. Where can you find the CarlaUE4.sh file ?
Thanks in advance

pardhuAj, 2020-12-07:
I tried using @agourishetty93's solution. It worked and scripts seems to be running fine

Mano2610, 2021-03-04:
@nagasanjay did the issue get resolved?

I am also getting the same error in windows. Kindly share me how to resolve this port issue.

ishaan95, 2021-06-21:
To start the server, you need to `make launch` and click Play. If the map is loaded properly, you should be able to move the camera around while in play mode. Have you tried opening a new terminal after clicking on Play and then running the example scripts?

closed 2023-06-02 by stale[bot]