carla-simulator/carla #3194: [Python API Segfault] sensor.other.lane_invasion causes crash when using qt matplotlib backend

hh0rva1h, 2020-08-16:
We have been tracking down a cumbersome python segfault that happens when using the matploblib qt backend together with sensor.other.lane_invasion.

The following file 100% reproduces the issue for us on different machines running different Linux / Python distributions (simply starting carla an executing the python file leads to an segfault of the python interpreter):

```
import matplotlib
import matplotlib.pyplot as plt
matplotlib.use("Qt5Agg")
#position of plt.figure() does not seem to matter, but the script does not crash if it is commented out!
plt.figure()

import carla

def _update_data(self, data):
    print("received data")

client = carla.Client("localhost", 2000)
world = client.load_world("Town04")

blueprints = world.get_blueprint_library()
default_car = blueprints.find("vehicle.tesla.model3")

car_position = carla.Transform(carla.Location(x=-10, y=16.5, z=12),
                                    carla.Rotation(yaw=180))

vehicle = world.spawn_actor(default_car, car_position)

env_lane_sensor = world.spawn_actor(blueprints.find("sensor.other.lane_invasion"),
                                            carla.Transform(), attach_to=vehicle)

print("Loaded everything, crash should happen now!")
env_lane_sensor.listen(_update_data)
print("Crash failed to trigger, are you using qt matplotlib backend?")
```

When commenting out `plt.figure()` the python interpreter segfault does not happen. Switching matplotlib backend to TkAgg for example is a workaround, still this should be investigated. Two members of our team hit that bug independently since the qt matplotlib backend is default. Tracking this down cost us hours in two occasions so we hope this can be investigated to save others from that pain.

We obtained the following backtrace from the python interpreter crash using apport in Ubuntu:
https://gist.github.com/hh0rva1h/b3e355fe4934b0167fcef8bc5c56f8fd

@germanros1987 Could you be so kind to label this issue accodringly? Thanks very much!

germanros1987, 2020-08-29:
@hh0rva1h are you getting this issue with a binary release or building CARLA from source? Could you provide the version? My intuition tells me this is ABI related (as usual).

@bernatx could you take a look at this?

hh0rva1h, 2020-08-29:
@germanros1987 Binary release, Carla 0.9.9.4.

hh0rva1h, 2020-11-04:
Still an issue.

closed 2021-04-27 by stale[bot]