carla-simulator/carla #3195: Clang version issues -

Mohith22, 2020-08-17:
I have clang-12 installed.  When I run "make launch", I get this error

clang 8 is required, but it's not installed.
make: *** [setup] Error 1

I am doing this on the university's machine which has CentOS and since I do not have root permissions, I build it from scratch.  
I followed steps given here - https://clang.llvm.org/get_started.html

But after installing clang this way, clang-12 got installed.  Is there a way to resolve this issue?

qhaas, 2020-08-18:
CARLA is [hard-coded](https://github.com/carla-simulator/carla/blob/0.9.9.4/Util/BuildTools/Setup.sh#L7) to use the file-path to clang-8 in a Debian/Ubuntu environment.  Without root permissions, you are looking at modifying CARLA's build scripts or using rootless podman/singularity Ubuntu containers to build.

We've grepped for occurrences of `/usr/bin/clang-8` in the build scripts and changed them manually.  However, using anything other than the version of llvm/clang tested by the CARLA developers might result in unexpected behavior.  Since you are in an Enterprise Linux (EL) environment, you will also be chasing down checks for [Debian-like environments](https://github.com/carla-simulator/carla/blob/0.9.9.4/PythonAPI/carla/setup.py#L38).

If you are on an EL7 x86-64 machine, I suspect the reason you are building from source is to avoid [glibc errors](https://github.com/carla-simulator/carla/issues/24) with the vendor provided binaries.  This can be mitigated by using a rootless [carla container](https://hub.docker.com/r/carlasim/carla) built by [singularity](https://sylabs.io/guides/3.6/user-guide/singularity_and_docker.html#building-images-for-singularity-from-docker-registries).

germanros1987, 2020-08-29:
@qhaas provided a great answer (thanks!). @Mohith22 indeed, the rootless docker container is your best option in this case. Could you please give it a try?

pengyuan-zhou, 2021-01-27:
> CARLA is [hard-coded](https://github.com/carla-simulator/carla/blob/0.9.9.4/Util/BuildTools/Setup.sh#L7) to use the file-path to clang-8 in a Debian/Ubuntu environment. Without root permissions, you [...]

Thanks @qhaas 
Manually changing clang version in the build scripts worked.
Ubuntu 20.04, clang-9, carla master branch.

closed 2021-07-21 by stale[bot]