carla-simulator/carla #3196: Error: Disabling core dumps was thrown while running "./CarlaUE4.sh"

tinmodeHuang, 2020-08-17:
**Issue:** the following error was thrown and that window still showd up  while running "./CarlaUE4.sh"
**Configures:**
Ubuntu 18.04 LTS desktop
GTX 1660Ti 6G
Nvidia-driver-440
opengl4.6
Carla 0.9.9.4 release

**Error:**
`4.24.3-0+++UE4+Release-4.24 518 0
Disabling core dumps.
FUnixPlatformMisc::RequestExitWithStatus
FUnixPlatformMisc::RequestExit
LowLevelFatalError [File:Unknown] [Line: 102] 
Exception thrown: bind: Address already in use
Signal 11 caught.
Malloc Size=65538 LargeMemoryPoolOffset=65554 
CommonUnixCrashHandler: Signal=11
Malloc Size=65535 LargeMemoryPoolOffset=131119 
Malloc Size=148816 LargeMemoryPoolOffset=279952 
Engine crash handling finished; re-raising signal 11 for the default handler. Good bye.
Segmentation fault (core dumped)`

**The ways (tried) & Corresponding results:**

1. ./CarlaUE4.sh followed by the option -opengl
no use

2. install Vulkan and its dependencies, but running vulkaninfo works makes it
no effect

3. Change it for previous one such as 0.9.5 release
no use

4. Be trying to downgrade graphics driver version from the current version 440
To be determined

5. There are some ways to be tried

nagasanjay, 2020-08-18:
I have the same issue. Also the window with a car doesn't show up. only one window without car (looks like an FPP of a drone) opens up.
[image]

revikiran1991, 2020-08-20:
I built the new master branch on Ubuntu 18.04 and packaged it. It also throws the same error. 

`4.24.3-0+++UE4+Release-4.24 518 0
Disabling core dumps.
X Error of failed request:  BadDrawable (invalid Pixmap or Window parameter)
  Major opcode of failed request:  149 ()
  Minor opcode of failed request:  4
  Resource id in failed request:  0x7c0003e
  Serial number of failed request:  365
  Current serial number in output stream:  376
terminating with uncaught exception of type std::__1::system_error: mutex lock failed: Invalid argument
Signal 6 caught.
Segmentation fault (core dumped)`

GPU: 1060 6GB

I tried running /PythonAPI/utils/config.py 

`  File "config.py", line 307, in <module>
    main()
  File "config.py", line 192, in main
    client = carla.Client(args.host, args.port, worker_threads=1)
AttributeError: module 'carla' has no attribute 'Client'
`

germanros1987, 2020-08-29:
@tinmodeHuang sorry for the delay. 

As it has been mentioned already, the "disabling core dumps" message is just letting you know that CARLA won't generate a core dump. However, I understand from your message that you are having problems running the server or client (this part wasn't clear to me).

In order to help you, could you please provide us with more information? First, for simplicity, we are going to consider CARLA 0.9.9.4 and vulkan (-vulkan).

* Are you running the server in a local machine connected to a physical screen?
* Does the server window show up? Can you interact with it?
* If you run the manual_control.py script, does it work?

tinmodeHuang, 2020-08-29:
> @tinmodeHuang sorry for the delay.
>
> As it has been mentioned already, the "disabling core dumps" message is just letting you know that CARLA won't generate a core dump. However, I understand from [...]

I won't modify posted description of the issue anymore in order to help those being in the same situation realise that message isn't a matter.
my answer to the first two questions is "yes",

RobinC94, 2020-09-29:
I have met the same problem with released pack carla0.9.6, which prints
`
4.22.1-0+++UE4+Release-4.22 517 0
Disabling core dumps.
sh: xdg-user-dir: command not found
Signal 11 caught.
Malloc Size=65538 LargeMemoryPoolOffset=65554 
CommonUnixCrashHandler: Signal=11
Malloc Size=65535 LargeMemoryPoolOffset=131119 
Malloc Size=44304 LargeMemoryPoolOffset=175440 
Engine crash handling finished; re-raising signal 11 for the default handler. Good bye.
`
For some reason, we are not able to run docker on server, so we have tried several machines we can borrow. We can only success with a Ubuntu16 server with 4 V100 GPUs. On the server we are using which is a Centos 7 server, it always return the above message. We have tried `-opengl`, `-opengl3` and even `-no-rendering` and nothing changes.
All these server has a GPU driver version 418.67 and cuda10.1
Does anyone knows what we can do next to make this issue fixed? We can do nothing because there are no coredump file to debug.
Many thanks!

tinmodeHuang, 2020-09-30:
> I have met the same problem with released pack carla0.9.6, which prints
> `4.22.1-0+++UE4+Release-4.22 517 0 Disabling core dumps. sh: xdg-user-dir: command not found Signal 11 caught. Malloc [...]

perhaps I couldn't have make it clear in the last comment, which the exception doesn't matter as germanros1987 said, thus you can open another terminal to run something relevant.

closed 2020-11-04 by tinmodeHuang

debanik123, 2021-05-09:
use that
./CarlaUE4.sh -opengl

honghua1998, 2021-07-14:
> use that
> ./CarlaUE4.sh -opengl
 
yes ! it works.but why does './CarlaUE4.sh'  fail to present the CarlaUE4 window continuesly?
thank you for your time

kartikeyshaurya, 2021-08-05:
> use that
> ./CarlaUE4.sh -opengl

worked for me

Derekabc, 2021-08-20:
[image]
I am still facing this problem and cannot figure out the reason!

WangTianYuan, 2021-09-10:
@Derekabc Have you solved the problem? I have met the same problem and I find it has no error information. So I don't know where to fix

pritesh604, 2022-02-08:
@tinmodeHuang  Have you found any solution because I am also facing the same issue.

Inglam, 2022-02-12:
@tinmodeHuang Have you found any solution because I am also facing the same issue. Thanks!

NaveedBhuiyan, 2023-02-03:
I have tried all the above but still cant find the solution. I am trying to run the Docker container for carla 9.13 on a windows 10 host. Maybe the issue is that I am using windows?

KangChou, 2023-04-19:
./CarlaUE4.sh -opengl -carla-server -benchmark -fps=10

Andrea010725, 2024-11-27:
Have you solved this problem? I'm having the same problem.

lvxiangyu11, 2024-12-11:
> 使用 ./CarlaUE4.sh -opengl

works to me, thx

FunKeen, 2025-10-23:
If you are using the method provided by Bench2Drive to install carla, then congratulations, the installation is complete, because the Bench2Drive author said
```
4.26.2-0+++UE4+Release-4.26 522 0 
Disabling core dumps. 
```
Only showing these two lines without termination is good.

mx2013713828, 2025-12-01:
> If you are using the method provided by Bench2Drive to install carla, then congratulations, the installation is complete, because the Bench2Drive author said
> 
> ```
> 4.26.2-0+++UE4+Release-4.26 522 0 
> Disabling core dumps. 
> ```
> 
> Only showing these two lines without termination is good.

什么意思大佬，我是按照这bench2drive安装的，也是这个报错，这样是没问题的吗？

Phoolore, 2026-02-17:
hi there. ChatGPT said I need vulcan
after
```
sudo apt install vulkan-tools
```
check with
```
vulkaninfo | less
```
and it worked, you can also try doing so with -nosound
```
./CarlaUE4.sh --quality-level=Low -nosound
```