carla-simulator/carla #3198: lens camera default parameters

dnns92, 2020-08-18:
Hey,

It's stated here (https://carla.readthedocs.io/en/latest/ref_sensors/#rgb-camera) that the default camera lens distortion parameters are non-zero, e.g.` lens_circle_falloff= 5.0` and `lens_k=-1.0`

Zero-ing every parameter leads to a single color screen as stated here: (https://github.com/carla-simulator/carla/issues/2822#issuecomment-653442599).

Now to my issue: Do the default lens distortion values lead to an undistorted pinhole-camera image?

If not, which parameters do lead to an undistorted image?

Cheers

marcgpuig, 2020-08-18:
> Do the default lens distortion values lead to an undistorted pinhole-camera image

Yes, @dnns92, as my comment you referenced says, default camera lens values lead to undistorted images.
Cheers!

closed 2020-08-29 by germanros1987

Rick831022, 2021-02-24:
> > Do the default lens distortion values lead to an undistorted pinhole-camera image
> 
> Yes, @dnns92, as my comment you referenced says, default camera lens values lead to undistorted images.
> Cheers!

Could you please give the description of the attribute? it's like a puzzle to guess the meaning of such as lens_k?