carla-simulator/carla #3200: Windows - Map ingestion to pre-build Carla

TKeutgens, 2020-08-18:
If have generated a custom map through RoadRunner and the map ingestion process from the build from source tutorial. (https://carla.readthedocs.io/en/latest/tuto_A_add_map/#map-ingestion-in-a-build-from-source)

In the build from source Carla version the map works well, but when I'm trying to load the map in my prebuild Carla version through `python config.py -m own_map` only the standard Carla map gets reloaded. Is there a possibility to load a custom build map for pre-build Carla?

germanros1987, 2020-08-29:
Did you have a look at this? https://carla.readthedocs.io/en/latest/tuto_A_add_map/#map-ingestion-in-a-carla-package

TKeutgens, 2020-09-01:
@germanros1987 until now I only tested the map creation through the build Carla version. I will try the Docker version. But it would be nice if both the Docker and the self build version would allow to share the created maps to a prebuild Carla.

TKeutgens, 2020-09-30:
@germanros1987  Unfortunately I was still not able to create a working Docker image. But I did a couple more tests on the self build version. I'm able to view and play the self built map in the Unreal Engine Editor. I was also able to create the .uasset for the map. 

In the documentation that you referenced there seems to be an error for the ingestion into the prebuild Windows Version of Carla. From my experience you have to place the maps in the WindowsNoEditor\CarlaUE4\Content folder, for <python config.py --list> to detect the map.

Strangely if I load the self build map with python <config --map my_own_map> no error gets thrown, but the ownly thing that happends is that the standard map gets reloaded. Is there any way to debug the map loading process?

I'm very thankfull for all hints!

niyantkumar, 2020-10-29:
I have generated a custom map using the docker image option and I have also kept my map into the WindowsNoEditor\CarlaUE4\Content/map_package folder. And the map is also shown as an available map when I run "client.get_available_maps()" command. But, when I try to load the map, it shows black screen, while the pre-built maps are working fine.

Also, it is a very simple map with two streets crossing and traffic lights at the junction. I made this map in the RoadRunner and followed each step fro ingesting this map into carla. 

Can someone please help????

EsquilliDev, 2021-01-13:
@niyantkumar Were you able to solve the black screen? I was/am experiencing the same issue.

germanros1987, 2021-01-26:
@TKeutgens any updates?

TKeutgens, 2021-01-27:
I tried adding the BP_Sky Object in the Unreal Editor. And there the map works fine. 

Then I tried moving the map package from 
`carla\Unreal\CarlaUE4\Content`
to a pre-build Carla. There the map is shown in the available maps. But each time I try to load the map, the default map (Town 03) gets reloaded. 

My idea was, that maybe there is a way to import a RoadRunner map into Carla on a Windows pc and to create a map package out of it to share with pre-build Carla users. Unfortunately I'm not familiar with the Unreal Engine.

closed 2021-07-21 by stale[bot]