carla-simulator/carla #3205: ImportError: libproj.so.9: cannot open shared object file. Precompiled RSS Version 0.9.9.4

leodario7, 2020-08-19:
I am trying to use the prebuild 0.9.9.4 version with RSS on a Ubuntu 18.04 machine, with an anaconda environment for python 3.7 with numpy and pygame already installed. 

```
>>> sys.path.append(glob.glob('../carla/dist/carla-*%d.%d-%s.egg' % (
...         sys.version_info.major,
...         sys.version_info.minor,
...         'win-amd64' if os.name == 'nt' else 'linux-x86_64'))[0])
>>> import carla
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "../carla/dist/carla-0.9.9-py3.7-linux-x86_64.egg/carla/__init__.py", line 8, in <module>
  File "../carla/dist/carla-0.9.9-py3.7-linux-x86_64.egg/carla/libcarla.py", line 7, in <module>
  File "../carla/dist/carla-0.9.9-py3.7-linux-x86_64.egg/carla/libcarla.py", line 6, in __bootstrap__
  File "/home/user/anaconda3/envs/carla/lib/python3.7/imp.py", line 342, in load_dynamic
    return _load(spec)
ImportError: libproj.so.9: cannot open shared object file: No such file or directory
```

This problem seems very similar to a possible bug reported and issued in #2871

germanros1987, 2020-08-29:
@fpasch could you please take a look?

fabianoboril, 2020-09-01:
Hi,

it seems proj4 (libproj) is not installed on your system, or in a different version. Ubuntu 18.04 by default ships with libproj.12.so. We have a PR open on the map repository, to address this issue, however it will take a bit until we have this in CARLA. So currently, you have to options. Either use Ubuntu 16.04 (docker container?) or install libproj.9.so by using the corresponding github version (https://github.com/OSGeo/PROJ/tree/da2d678c387ba76704824360dc52771c29d94693) and a manual build+install.

YukunXia, 2020-09-03:
> Hi,
>
> it seems proj4 (libproj) is not installed on your system, or in a different version. Ubuntu 18.04 by default ships with libproj.12.so. We have a PR open on the map repository, to address this [...]

Hi, this Github version of proj, tagged as "tags/4.9.3", installs libproj.12.so on my ubuntu 20.04, rather than libproj.9.so.

closed 2020-11-14 by stale[bot]