carla-simulator/carla #3208: Is there a script like view_start_positions.py for Carla 0.9 and up?

pparmesh, 2020-08-20:
I tried running view_start_positions.py from an older version (0.8.2) but the client couldn't find the localhost so I'm guessing it might be a compatibility issue. Basically I'm trying to visualize spawn points as I want to spawn my vehicle at a particular location in a map.

germanros1987, 2020-08-29:
@pparmesh in newer versions of CARLA (0.9.8, 0.9.9, etc.) you can spawn vehicles anywhere as long as there are no collisions. In any case, you can visualize spawn points using the non_rendering_mode.py with --show-spawn-points

closed 2020-08-29 by germanros1987

RichardJimenezG, 2024-06-06:
@germanros1987 how is it possible to achieve visualizing spawn points, non_rendering_mode.py  doesn't  seem to be available anymore in even newer versions CARLA (0.9.14, 0.9.15)?
Many thanks in advance.