carla-simulator/carla #3209: How to inspect cpp code in LibCarla when debugging from python script

youngsend, 2020-08-21:
Hello, @nsubiron !
I encountered an error when running spawn_npc.py.
```bash
$ python3 spawn_npc.py
ERROR: Spawn failed because of collision at spawn position
ERROR: Spawn failed because of collision at spawn position
ERROR: Spawn failed because of collision at spawn position
ERROR: Spawn failed because of collision at spawn position
ERROR: Spawn failed because of collision at spawn position
ERROR: Spawn failed because of collision at spawn position
ERROR: Spawn failed because of collision at spawn position
ERROR: Spawn failed because of collision at spawn position
ERROR: Spawn failed because of collision at spawn position
ERROR: Spawn failed because of collision at spawn position
ERROR: Spawn failed because of collision at spawn position
ERROR: Spawn failed because of collision at spawn position
ERROR: Spawn failed because of collision at spawn position
ERROR: Spawn failed because of collision at spawn position
terminate called after throwing an instance of 'std::out_of_range'
  what():  vector::_M_range_check: __n (which is 0) >= this->size() (which is 0)
Aborted (core dumped)
```
I want to inspect where this error comes from, so I ran it again in gdb.
However, the information did not change, and I still cannot locate the cpp source file.
Could you please tell me how to debug into LibCarla cpp file and see values of variables when this error happens?
(I'm sorry that this may not be a carla issue.)
I am using carla 0.9.8.
```bash
(gdb) run spawn_npc.py 
Starting program: /usr/bin/python3 spawn_npc.py
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7ffff38d2700 (LWP 18549)]
[New Thread 0x7ffff30d1700 (LWP 18550)]
[New Thread 0x7ffff28d0700 (LWP 18551)]
[New Thread 0x7ffff20cf700 (LWP 18552)]
[New Thread 0x7ffff18ce700 (LWP 18553)]
[New Thread 0x7ffff10cd700 (LWP 18554)]
[Thread 0x7ffff10cd700 (LWP 18554) exited]
[New Thread 0x7ffff08cc700 (LWP 18555)]
[New Thread 0x7ffff10cd700 (LWP 18556)]
[Thread 0x7ffff08cc700 (LWP 18555) exited]
[New Thread 0x7ffff08cc700 (LWP 18578)]
[New Thread 0x7fffd9ffe700 (LWP 18579)]
[New Thread 0x7fffd97fd700 (LWP 18580)]
[New Thread 0x7fffd8ffc700 (LWP 18581)]
[New Thread 0x7fffd093b700 (LWP 18582)]
[New Thread 0x7fffdbfff700 (LWP 18583)]
[Thread 0x7fffdbfff700 (LWP 18583) exited]
[New Thread 0x7fffdbfff700 (LWP 18584)]
[Thread 0x7fffdbfff700 (LWP 18584) exited]
[New Thread 0x7fffdb7fe700 (LWP 18585)]
ERROR: Spawn failed because of collision at spawn position
ERROR: Spawn failed because of collision at spawn position
ERROR: Spawn failed because of collision at spawn position
ERROR: Spawn failed because of collision at spawn position
ERROR: Spawn failed because of collision at spawn position
ERROR: Spawn failed because of collision at spawn position
ERROR: Spawn failed because of collision at spawn position
ERROR: Spawn failed because of collision at spawn position
ERROR: Spawn failed because of collision at spawn position
ERROR: Spawn failed because of collision at spawn position
ERROR: Spawn failed because of collision at spawn position
ERROR: Spawn failed because of collision at spawn position
ERROR: Spawn failed because of collision at spawn position
ERROR: Spawn failed because of collision at spawn position
ERROR: Spawn failed because of collision at spawn position
[Thread 0x7fffdb7fe700 (LWP 18585) exited]
[New Thread 0x7fffdb7fe700 (LWP 18586)]
[Thread 0x7fffdb7fe700 (LWP 18586) exited]
spawned 10 vehicles and 35 walkers, press Ctrl+C to exit.
terminate called after throwing an instance of 'std::out_of_range'
  what():  vector::_M_range_check: __n (which is 0) >= this->size() (which is 0)

Thread 10 "python3" received signal SIGABRT, Aborted.
[Switching to Thread 0x7ffff08cc700 (LWP 18578)]
0x00007ffff7825438 in __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:54
54	../sysdeps/unix/sysv/linux/raise.c: No such file or directory.
(gdb) bt
#0  0x00007ffff7825438 in __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:54
#1  0x00007ffff782703a in __GI_abort () at abort.c:89
#2  0x00007ffff505c835 in __gnu_cxx::__verbose_terminate_handler() () from /home/sen/.python-eggs/carla-0.9.8-py3.5-linux-x86_64.egg-tmp/carla/libcarla.cpython-35m-x86_64-linux-gnu.so
#3  0x00007ffff4fc8cf6 in __cxxabiv1::__terminate(void (*)()) () from /home/sen/.python-eggs/carla-0.9.8-py3.5-linux-x86_64.egg-tmp/carla/libcarla.cpython-35m-x86_64-linux-gnu.so
#4  0x00007ffff4fc8d41 in std::terminate() () from /home/sen/.python-eggs/carla-0.9.8-py3.5-linux-x86_64.egg-tmp/carla/libcarla.cpython-35m-x86_64-linux-gnu.so
#5  0x00007ffff4ff655e in execute_native_thread_routine () from /home/sen/.python-eggs/carla-0.9.8-py3.5-linux-x86_64.egg-tmp/carla/libcarla.cpython-35m-x86_64-linux-gnu.so
#6  0x00007ffff7bc16ba in start_thread (arg=0x7ffff08cc700) at pthread_create.c:333
#7  0x00007ffff78f74dd in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:109
(gdb) up
```

_Originally posted by @youngsend in https://github.com/carla-simulator/carla/issues/2747#issuecomment-678027425_

thytless, 2020-08-21:
As far as I know, you can make debug print (std::cout or what)  in Libcarla cpp code,  and debug info will be available after recompiling PythonAPI(make PythonAPI.3).

youngsend, 2020-08-25:
I succeeded in debugging it using `catch throw` command provided by gdb.

closed 2020-08-25 by youngsend