carla-simulator/carla #3216: Error when running "make package"

RicardoSakura, 2020-08-24:
I am using Carla 0.9.9. I have built the Unreal engine successfully, but got an error with the command of
make package in Ubuntu 18.0.4.
[errorlog.txt](https://github.com/carla-simulator/carla/files/5115482/errorlog.txt)

germanros1987, 2020-08-29:
@RicardoSakura could you please provide more details about your setup?
@sergi-e could you try to help?

sergi-e, 2020-08-31:
Hi @RicardoSakura!

It would be great if you could provide us with your log, so that we can track the error :)

rebolomo, 2020-10-10:
I have the same issue when making package with carla 0.9.9. Any help will be appreciated.
[log.txt](https://github.com/carla-simulator/carla/files/5358602/log.txt)
[Cook-2020.10.10-15.36.43.txt](https://github.com/carla-simulator/carla/files/5358603/Cook-2020.10.10-15.36.43.txt)

closed 2020-12-25 by stale[bot]