carla-simulator/carla #3217: CARLA Cannot see Vehicle after spawning

souvikboseGM, 2020-08-24:
Hi,

I am new to CARLA [Using CARLA 0.9.8 ,Ubuntu 18]. Any help is appreciated

I tried to load a Town5 and wanted to drive the vehicle Automatically. Town is loaded .But when I spawn the vehicle with "vehicle = world.spawn_actor(bp, transform)" , I see 2 issue

1> I could not see the vehicle on the Road .
2> And also I am getting the error [error msg atatched]

[image]

Below is the Code : I am running the code by -> python ATW_Souvik.py>

import glob
import os
import sys

import carla
import random
import time

def main():
actor_list = []
try:

    # requests in the localhost at port 2000.
    client = carla.Client('localhost', 2000)
    client.set_timeout(60)

    world = client.get_world()
    world = client.load_world('Town05')    
    blueprint_library = world.get_blueprint_library()
    bp = random.choice(blueprint_library.filter('vehicle'))
    transform = random.choice(world.get_map().get_spawn_points())
    vehicle = world.spawn_actor(bp, transform)
    actor_list.append(vehicle)
    

    # Let's put the vehicle to drive around.
    vehicle.set_autopilot(True)
    time.sleep(5)

finally:

    print('destroying actors')
    for actor in actor_list:
        pass
        actor.destroy()
    print('done.')
if name == 'main':

main()

jackbart94, 2020-08-24:
Hi,

You're not seeing it because it gets destroyed after 5 seconds. The error is fixed in 0.9.10 and it's related to the actor.destroy() function, but it still works indeed.

souvikboseGM, 2020-08-24:
Thanks Jack for the reply. I was thinking there should definitely be way/logic to keep continuing in simulation Mode with 0.9.8 ?

closed 2020-08-24 by souvikboseGM