carla-simulator/carla #3218: .osm file to .xodr   [AttributeError: module 'carla' has no attribute 'Osm2OdrSettings']

souvikboseGM, 2020-08-24:
Hi , Goodmorning,

I was trying to convert .osm file to .xodr file to use in CARLA [Using CARLA 0.9.8 ,Ubuntu 18].
I am getting error message = > " AttributeError: module 'carla' has no attribute 'Osm2OdrSettings' "
I am not sure how to get this "Osm2OdrSettings" Module ? 

Below is the  code .  

import carla
# Read the .osm data
f = open("map_openstreet.osm", 'r')
osm_data = f.read()
f.close()

# Define the desired settings. In this case, default values.
settings = carla.Osm2OdrSettings()
# Convert to .xodr
xodr_data = carla.Osm2Odr.convert(osm_data, settings)

# save opendrive file
f = open("CARLMAP_openstreet.xodr", 'w')
f.write(xodr_data)
f.close()

TKeutgens, 2020-08-25:
@souvikboseGM this function is only available in the Pre-Release 0.9.10 Version -> https://github.com/carla-simulator/carla/blob/master/Docs/download.md#pre-release-0910

souvikboseGM, 2020-08-26:
Thanks for the reply. I downloaded  "Pre-Release 0.9.10 Version "  and tried to run. 
I got the error msg => "ImptortError: libxerces-c-3.1.so: cannot open shared object file: No such file or directory"

I tried to fix the above error by 
sudo apt-get update
sudo apt-get install libxerces-c3.1

Then I got below error.  Not sure ,how to proceed to next step . Any help is appreciated ?
[image]

souvikboseGM, 2020-08-26:
@TKeutgens Thanks for the reply. After downloading "Pre-Release 0.9.10 Version", I tried to run. Now I am getting below error
I am not sure ,how to go about it ?

[image]

TKeutgens, 2020-08-26:
@souvikboseGM I'm not 100% sure what the error is but you can try running your script from the folder WindowsNoEditor\PythonAPI\examples. For me that worked. 

The xsd file sould be located under WindowsNoEditor\PythonAPI\carla\data\xsd

souvikboseGM, 2020-08-26:
@TKeutgens It works . Great. I am getting below warnings .I am assuming this should not be a problem. If you could point out something about "junctions.scurve-stretch",please let me know. In any case , .xodr is generated, so I will close this issue. But if you could point something about the warnings that will be great. Thanks again

[image]

closed 2020-08-26 by souvikboseGM

salvatoresan, 2020-08-27:
Hi @souvikboseGM, I'm also trying to convert the osm data but i'm getting the following error message when i try to run the converter script inside the PythonAPI/examples folder: 
`AttributeError:` 'module' object has no attribute 'Osm2OdrSettings'
Could you tell me the exact steps that you followed? In which folder did you place the osm data?
 I downloaded the pre-release version and used the same code suggested in the documentation
Thanks

souvikboseGM, 2020-08-27:
@salvatoresan 
I used the below code and running the below code from  "..PythonAPI/examples" path

Read the .osm data
f = open("map_openstreet.osm", 'r')
osm_data = f.read()
f.close()

Define the desired settings. In this case, default values.
settings = carla.Osm2OdrSettings()

Convert to .xodr
xodr_data = carla.Osm2Odr.convert(osm_data, settings)

save opendrive file
f = open("CARLMAP_openstreet.xodr", 'w')
f.write(xodr_data)
f.close()