carla-simulator/carla #3219: control multi vehicles

carlaUE4new, 2020-08-24:
I am trying to control multiple vehicles seperately within the same world. Does Carla 0.9.9.4 support this? if supported, is there an example of how to do this? Do I need to set up each vehicle will communicate with world through different TCP ports?

Thanks

thytless, 2020-08-25:
I believe that the answer is yes. Actually client sends control command to the server, and that is how vehicles are controlled. Prepare control commands for each actor. Learn more about it: https://carla.readthedocs.io/en/latest/python_api/#commandapplyvehiclecontrol

germanros1987, 2020-08-29:
Indeed. The API lets you control as many vehicles as you need. You just need to follow the instructions, as indicated by @thytless

closed 2020-08-29 by germanros1987