carla-simulator/carla #3220: apply_settings not returning after loading a new world

dHonerkamp, 2020-08-25:
Hi,

I am running code in the form of
```
      client = carla.Client("localhost", port)  # pylint: disable=no-member
      client.set_timeout(20)
      client.load_world(map_name="Town01")
      world = client.get_world()
      world.set_weather(carla.WeatherParameters.ClearNoon)  # pylint: disable=no-member
      world.wait_for_tick()
      frame = world.apply_settings(
          carla.WorldSettings(  # pylint: disable=no-member
              no_rendering_mode=False,
              synchronous_mode=True,
              fixed_delta_seconds=1.0 / fps,
          ))
```

Where after some time I want to load a new world (using the same code).
The problem is that if in between I spawn vehicles, then when running this for the second time, the `apply_settings` call never returns. This only happens when spawning vehicles, not when using an empty map or spawning only pedestrians.

```
     client.load_world(map_name="Town01")
     world = client.get_world()
      frame = world.apply_settings(
          carla.WorldSettings(  # pylint: disable=no-member
              no_rendering_mode=False,
              synchronous_mode=True,
              fixed_delta_seconds=1.0 / 20,
          ))
```
Note that I can successfully retrieve the new map after loading it, as well as do things like applying a new wheather condition etc.. It is only (re-)applying the setttings that fails.

I've tried to first ensure that all actors are destroyed (using the code below), but that didn't make any difference. I've also tried to first do `world.wait_for_tick()` before applying the settings. Again to no effect.
```
self._client.apply_batch([carla.command.DestroyActor(x) for x in [self._hero] + self._vehicles + self._pedestrians])
```

Am I missing a required step to reset the world? (The documentation seems to state that loading a new world automatically takes care of everything).
```
Client API version     = 0.9.9.4-255-gc56a7738
Simulator API version  = ba812e75  (latest docker image)
```

dHonerkamp, 2020-08-28:
It seems that the problem is actually the traffic manager, chaning the title accordingly.
As soon as I set the synchronous mode on the traffic manager, the world.apply_settings() call described above no longer returns after loading or reloading a world. I.e. after doing this:
```
    traffic_manager = client.get_trafficmanager(8000)
    traffic_manager.set_synchronous_mode(True)
```
This gets stuck in the apply_settings() call:
```
     client.reload_world()
     world = client.get_world()
      frame = world.apply_settings(
          carla.WorldSettings(  # pylint: disable=no-member
              no_rendering_mode=False,
              synchronous_mode=True,
              fixed_delta_seconds=1.0 / 20,
          ))
```

I have tried to set the synchronous mode to false again before reloading a world, but that did not help either. The documentation (https://carla.readthedocs.io/en/latest/adv_traffic_manager/) states that I do not have to destroy the traffic manager myself.

Any help would be much appreciated.

germanros1987, 2020-08-29:
@jackbart94 could you please take a look at this?

Beskhue, 2020-09-02:
This might be related to #2789.

adscib, 2020-09-02:
My guess would be that the problem is entirely on the client side. If I'm running a client with a traffic manager connecting to a server in synchronous mode, then start another server, get a client for the second server from the Python process running the first client, load a world and try to apply synchronous settings for the second server, that last call doesn't return. The issue goes away if I load the world and apply settings on both servers before getting a traffic manager on either one. I therefore suspect that setting the traffic manager in synchronous mode leaves some piece of state on the client side that prevents the call applying synchronous settings on the world from returning.

dHonerkamp, 2020-09-03:
Makes sense, after restarting the python script (not the server) it does indeed work again, as also mentioned in the issue linked above.

Beskhue, 2020-09-09:
This is a possible workaround for this issue: https://github.com/carla-simulator/carla/issues/2789#issuecomment-689619998.

jackbart94, 2020-09-15:
Sorry if it took so much to open a PR for this issue, it appears we have found a way to fix it!

It will soon be merged in `dev` and then to `master` when 0.9.10 comes out, but if you want to test it (and give us feedback!) the branch is `traffic_manager/map_change_fix`

Thanks for your patience!

adscib, 2020-09-15:
Thanks for the fix! I can confirm that I'm now able to reload the world using the CARLA version from `traffic_manager/map_change_fix`.

PR #3288 (2020-09-16): Fixes a map change error when TM is in synchronous mode
#### Description

An erroneous update of the TM while in synchronous mode was causing the client to hang on map changes.

Fixes #3220
Fixes #2789

#### Where has this been tested?

  * **Platform(s):** Ubuntu 18.04, Windows
  * **Python version(s):** Python 3.6
  * **Unreal Engine version(s):** 4.24

#### Possible Drawbacks

Seems to not be working for Windows, further testing is required.

---
This change is [[image]](https://reviewable.io/reviews/carla-simulator/carla/3288)
closed 2020-09-16 by bernatx