carla-simulator/carla #3221: carla.VehiclePhysicsControl is missing moment of inertia and proper wheel positions

thomasfermi, 2020-08-25:
Today I used the carla.VehiclePhysicsControl python object and came across these problems:

1. carla.VehiclePhysicsControl has the attribute mass but not moment of inertia (there is the moment of inertia of the engine, but I mean the moment of inertia of the whole car)

2. carla.WheelPhysicsControl has an attribute position, but it is in centimers and not clear in which reference frame it is. It would be nice that it would be in meters and in the same reference frame as carla.VehiclePhysicsControl.center_of_mass. Regarding the center_of_mass, it is also not clear in which reference frame that is? Is it the one relative to vehicle.get_transform()?

germanros1987, 2020-08-29:
@DSantosO could you please account for these suggestions?

thomasfermi, 2020-11-03:
It seems that nobody found time to work on this yet. 

For people who need a solution now (which is quite ugly, I admit):
**Estimating distance between wheels:**
- You can attach a camera to the vehicle, move it sideways and rotate the yaw by 90°. Now you are looking at your vehicle from the side. Move the camera forwards by x meters and check if the front tire is at the center of the image. Tune x until the front tire is at the image center. Do the same for the rear tire by moving the camera y meters backwards. The distance between the wheels is x+y

**Estimating moment of inertia:**
Empirically
I_zz = 0.95 M*(L/2)^2
seems to be a good estimate for most cars (hopefully also in Carla). Here M=mass, L=wheelbase=distance front wheel to rear wheel
Source:
https://www.nhtsa.gov/DOT/NHTSA/NRD/Multimedia/PDFs/VRTC/ca/capubs/sae1999-01-1336.pdf

closed 2021-04-27 by stale[bot]