carla-simulator/carla #3222: Have problem in label <signals></signals> when import customize opendrive map

H-Whisky, 2020-08-26:
Platform: win10
Carla Version: 0.9.9
UE4 Engine Version: 4.24.3
Python Version: 3.7.3

Hi, From the official documentation, I can import the official and carla-provided opendrive files into UE4_Carla through the following script, but importing a custom opendrive file and parsing the <signals></signals> tag caused UE4 to crash and make an error.
The correct content is as follows(running is the TownBig.xodr file): 
[image]

The error content is as follows(running is the customize xodr file):
[image]
UE4 crashed directly！

Then I found that there was an error in the trafficlightmanager.cpp file, so I deleted all the signals tags in the custom xodr file.
Surprisingly, I can import a customized map into UE4, but unfortunately, all the signal lights and road signs have disappeared
The content is as follows(running is the customize xodr file without signals tags):
[image]

So I compared the official and my customized <signals> and found that there is a difference. Can you help me explain the difference and how to change the customized <signals> into an official similar and successful Import into UE4_carla：

my.xodr:
`
        <signals>
            <signal s="3.0000000000000000e+00" t="4.5999999999999996e+00" id="175" name="_Sg175" dynamic="yes" orientation="+" zOffset="2.9999999999999999e-01" type="1000002" country="OpenDRIVE" subtype="-1" hOffset="0.0000000000000000e+00" pitch="0.0000000000000000e+00" roll="0.0000000000000000e+00" height="2.62" width="0.35">
                <validity fromLane="0" toLane="0"/>
                <userData code="3" value="0.0000000000000000e+00"/>
            </signal>
            <signal s="3.0000000000000000e+00" t="-4.5999999999999996e+00" id="176" name="_Sg176" dynamic="yes" orientation="+" zOffset="2.9999999999999999e-01" type="1000002" country="OpenDRIVE" subtype="-1" hOffset="0.0000000000000000e+00" pitch="0.0000000000000000e+00" roll="0.0000000000000000e+00" height="2.62" width="0.35">
                <validity fromLane="0" toLane="0"/>
                <userData code="3" value="0.0000000000000000e+00"/>
            </signal>
        </signals>
`

TownBig.xodr:
`
        <signals>
            <signal name="Sign_Yield" id="6080" s="6.3103172793201823e-2" t="5.6349899716323364e+0" zOffset="2.5534391922906252e+0" hOffset="3.1415927410125732e+0" roll="7.2123139708765716e-13" pitch="3.8721170858480036e-4" orientation="+" dynamic="no" country="OpenDRIVE" type="205" subtype="-1" value="-1.0000000000000000e+0" text="" height="1.0000000000000000e+0" width="1.1252747774124146e+0">
                <validity fromLane="0" toLane="0"/>
                <userData>
                    <vectorSignal signalId="{424cac1f-bc26-4dcc-971b-faa881bb081c}"/>
                </userData>
            </signal>
        </signals>
`

germanros1987, 2020-08-29:
@Axel1092 this needs your attention.

Axel1092, 2020-09-04:
Hi @H-Whisky, thank you for your report, this is a problem that happens when a signal is having an effect in a lane that has 0 width which may happen when the lane is.

H-Whisky, 2020-09-08:
> If you work from source you may try to update the code, otherwise a work arround would be to make sure that signals are placed or references in lanes with width greater than 0.

Thanks @Axel1092! I will update the code in the current master branch. And try again.

H-Whisky, 2020-09-22:
> Hi @H-Whisky, thank you for your report, this is a problem that happens when a signal is having an effect in a lane that has 0 width which may happen at the start of a lane. This issue will be fixed in the next release and is fixed in the current master branch. If you work from source you may try to update the code, otherwise a work arround would be to make sure that signals are placed or references in lanes with width greater than 0.

Hi@Axel1092,  I meet the issues when I update the code to master. I can run "make launch" and run a simulator by the Unreal Edior. But when I run "make PythonAPI" to get the corresponding *.egg file, I meet the following error:
```
d:\workspace\carla\pythonapi\carla\source\libcarla\OSM2ODR.cpp(7): fatal error C1083: Cannot open include file: “OSM2ODR.h”: No such file or directory
error: command 'D:\\software\\VisualStudio2017\\VC\\Tools\\MSVC\\14.16.27023\\bin\\HostX86\\x64\\cl.exe' failed with exit status 2

-[BuildPythonAPI]: Carla lib for python has been successfully installed in "D:\workspace\carla\PythonAPI\carla\dist"!
```
I see the error notification, so I go to the carla/build, and I see that the **xerces-c-3.2.3-install** folder is null, and the **xerces-c-3.2.3-source** has been download, I don't know how to make the **xerces-c-3.2.3-install** really installded. Need you help. Thank you!

closed 2020-12-13 by stale[bot]