carla-simulator/carla #3224: Traffic Manager doesn't react on multiple speed changes

patrick-11, 2020-08-26:
Carla: 0.9.9.4
OS: Windows10
Python: 3.7.9

Hello, currently I am working with multiple (around 200) vehicles that are subscribed to one traffic manager with the autopilot on. In my project I need to change multiple times the speed of individual vehicles, but I noticed that sometimes the traffic manager doesn't react to speed changes. For example, in the code below the simulation reacts to the first speed change, but after the timeout the same vehicles won't speed up.

`tm = client.get_trafficmanager(8000)
for npc in npc_list:
    tm.vehicle_percentage_speed_difference(npc, 90.0)

time.sleep(10)

for npc in npc_list:
    tm.vehicle_percentage_speed_difference(npc, 30.0)`

Do I need to create batches and apply them with the "apply_batch_sync()" method? If so, how would that work? Or is the issue somewhere else?

germanros1987, 2020-08-29:
@jackbart94 could you please provide some feedback?

jackbart94, 2020-09-03:
Hi,

Sorry for the delay. Yes, we had faced this issue for this particular parameter. It's been fixed, a PR will be opened soon, thanks for opening this issue!

PR #3262 (2020-09-17): Fixes add entry issue for applying parameters more than once
#### Description

Changes in Traffic Manager to fix an issue related to applying specific parameters more than once in the same execution.

Fixes #3224
Fixes #2966

#### Where has this been tested?

  * **Platform(s):** Ubuntu 18
  * **Python version(s):** 3.6
  * **Unreal Engine version(s):** 4.24

#### Possible Drawbacks

None

---
This change is [[image]](https://reviewable.io/reviews/carla-simulator/carla/3262)
closed 2020-09-04 by patrick-11