carla-simulator/carla #3225: cannot open CarlaUE4 project with UE4 4.24.3

dingkailv, 2020-08-26:
I downloaded the carla 0.9.9 for windows. The binaries run perfectly, nothing weird happened. But when I want to open the CarlaUE4.uproject under CarlaUE4, it says the CarlaUE4 and Carla are missing or built with a different engine version. That I have to re-build them

I had this problem when trying to build Carla as well, I can build it. But cannot open the UE4 project.
What seems to be wrong?

Lu7913, 2020-08-28:
Same here, couldn't solve the issue, building Carla from source solved the issue for me.

germanros1987, 2020-08-29:
@dingkailv you can't build any version of CARLA with any version of UE4. Each version of CARLA is designed for a specific version of UE4.

closed 2020-08-29 by germanros1987