carla-simulator/carla #3229: manual_control.py exits after 2s without an error

xmgpev, 2020-08-29:
Hello,

I compiled CARLA 0.9.9.4 and PythonAPI (on Win10). Everything worked fine, the compilation completed without errors. I opened the CarlaUE4.uproject and clicked the play button. When I run the manual_control.py script, the pygame window only shows up for 2 seconds and after that it closes with no errors.

I debugged it and found that client.get_world() gave me an error message: 
`[WinError 10054] An existing connection was forcibly closed by the remote host.`

I have tried other (also the compiled) versions and the result is the same.

I hope someone can help me. Thanks in advance!

germanros1987, 2020-09-24:
Sorry for the delay. @jackbart94 could you provide some feedback about this issue?

xmgpev, 2020-10-08:
I finally found the bug in my case. I had several versions of Carla on my laptop and the system path variable CARLA_ROOT was configured for an older version of the PythonAPI.

closed 2020-10-08 by xmgpev