carla-simulator/carla #3231: can't get carlaviz in web

whuxhzhang, 2020-08-30:
I am trying use carlaviz, and ss of this step, it's normal: docker run -it --network="host" -e CARLAVIZ_HOST_IP=localhost -e CARLA_SERVER_IP=localhost -e CARLA_SERVER_PORT=2000 <name_of_Docker_image>
but when I type in the browser " http://127.0.0.1 :8080/ ", the output is not like the tutorial shows.
[image]

wx9698, 2020-08-31:
Make sure that your localhost's 8080 port is not bound by another process. And by the way what is ss?

whuxhzhang, 2020-08-31:
> Make sure that your localhost's 8080 port is not bound by another process. And by the way what is ss?

Sorry, it's my spelling mistake...

I'll check as you say，thank you very much.

whuxhzhang, 2020-09-01:
> Make sure that your localhost's 8080 port is not bound by another process. And by the way what is ss?

If  the host's 8080 port is bound by another process, can I change the carlaviz's port ？

wx9698, 2020-09-01:
If you want to specify the port, you need to build this project (at least fronend part) by yourself instead of running frontend in the docker container.

You need to modify the start command in the scripts section in file frontend/package.json. 
Add "--port PORT_YOU_WANT" in that command and run "yarn start".

I will add more configuration options in future releases.

whuxhzhang, 2020-09-01:
> If you want to specify the port, you need to build this project (at least fronend part) by yourself instead of running frontend in the docker container.
>
> You need to modify the start command in the [...]

It works!!!
Turn off the use of this port on my computer， it working.
Thank you very much.

closed 2020-09-01 by whuxhzhang