carla-simulator/carla #3232: Co-simulation problem with Sumo

zwbai, 2020-08-30:
Hi,

I want to try the co-simulation with sumo. Following the tutorial, I ran the following code:
`python run_synchronization.py examples/Town04.sumocfg  --sumo-gui`
and when I click the play button in sumo, it showed "simulation ended":
[image]

so, could anyone tell me what;s the problem and what should I do?

Thanks!

zwbai, 2020-08-30:
It said the reason is "TraCl requested termination", but I didn't find how to solve it.
 Here is the log:
[image]

joel-mb, 2020-09-01:
Hi @zwbai!
Which version of sumo are you using? You can check it by running `sumo --version`
Have you tested with other towns (e.g., Town01)?

zwbai, 2020-09-10:
Hi @joel-mb !

Thank you for your attantion. My sumo version is 0.32.0.
In addition, I also tried Town01 with same command, but it shows the same error, which is shown as below:

[image]

hope that you can help to solve the problem, thank you very much!

joel-mb, 2020-09-14:
Hi @zwbai  ! Are you using this version of SUMO for any specific reason? I would say that if you update SUMO to the latest version (at least to `1.5`) the issue should disappear.

zwbai, 2020-09-14:
It worked! Thank you very much.

HemaZ, 2020-09-15:
I'm still getting a similar issue running SUMO 1.6 
with `python spawn_npc_sumo.py --sumo-gui`
[image]

joel-mb, 2020-09-21:
Hi @HemaZ! Which version of CARLA are you using? This issue was fixed here: https://github.com/carla-simulator/carla/pull/3129

HemaZ, 2020-09-21:
> Hi @HemaZ! Which version of CARLA are you using? This issue was fixed here: #3129
Hi @joel-mb I'm using CARLA 9.9.4

joel-mb, 2020-09-25:
Hi @HemaZ !
CARLA 0.9.9.4 is not containing the fix for this issue. You should use CARLA 0.9.10 or just update you `spawn_npc_sumo` script by this one: https://github.com/carla-simulator/carla/blob/master/Co-Simulation/Sumo/spawn_npc_sumo.py

closed 2020-09-25 by joel-mb