carla-simulator/carla #3235: Can't find Visual Studio Compiler (cl.exe)

elifozisik, 2020-08-31:
Hey guys, first of all im not an expert so this might be an easy problem because i did not see any similar issues in forums but  really tried to fix this in many ways and couldn't fix.

So ,i'm installing CARLA for the first time and i did _almost_ everything on guide (except setting the environment variable because i didn't get where **"choose the path to the installation folder of the desire UE4 installation"** is). I can see **Generate Visual Studio project files** option on .uproject files, UE4.2 is working, i reinstalled VS17 and it is working. I extracted newest assets-0.9.10 to _content_ as described. Now for every make command im getting this errors. I added VS17 bin and cl.exe paths to PATH (which is in a weird folder _C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.16.27023\bin\Hostx64\x64_).

What am i missing?

[image]

closed 2020-09-08 by elifozisik

nagasanjay, 2020-10-08:
Hi.. I face the same issue... Can you share with me how did you solve it ?

elifozisik, 2020-10-16:
@nagasanjay hi sorry for late answer.
It was actually quite simple, i added system32 to my path. Apparently i deleted it before.
Adding C: \   **C:\windows\System32**   C:\Program Files solved my problem

EchoYJJ, 2020-12-20:
Hi @elifozisik when I run "win+R", my VS2017 Environment initialized for: 'x64_x86', not x64. Do you know how to change it?
[image]

elifozisik, 2020-12-20:
Hi @EchoYJJ  im not sure but maybe your path has VS x64_x86 and you can try changing it to your x64 adress. 
The variable in my path;
**C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.16.27023\bin\Hostx64\x64\cl.exe;**

Also when i write cmd in win+R, it doesn't show the x64 part in my command prompt, but when i manually search x64 it does show it. So im not sure, maybe you can search that **vcvarsall.bat** command

EchoYJJ, 2020-12-21:
@elifozisik Thanks a lot! It works for me.

arsalan0302, 2022-09-14:
@elifozisik 
Hello Sir!
I have the same issue you faced, we have tried to solve it but could not reach  the specific path.
Please, guide us step by step s much as possible

seominseok00, 2024-04-03:
Adding the following path to the environment variables resolved the issue.

C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30133\bin\Hostx64\x64

[image]