carla-simulator/carla #3236: carla-simulator : Depends: libjpeg8 but it is not installable

keli5, 2020-08-31:
OS: Debian 10.5 x86_64 (4.19.0-10-amd64)

I'm trying to install with the deb instructions in the quick start guide:
```
sudo apt-get update
sudo apt-get install carla-simulator
cd /opt/carla-simulator
```
However, after apt-get install carla-simulator, it fails with this error: 
```
The following packages have unmet dependencies:
 carla-simulator : Depends: libjpeg8 but it is not installable
E: Unable to correct problems, you have held broken packages.
```
After this, the `cd /opt/carla-simulator` fails as the directory doesn't exist.
Running `sudo apt-get install -f` doesn't help.

qhaas, 2020-09-04:
Similar issue in RHEL as described in Issue #1783.  IIRC, the current version of CARLA targets Ubuntu 18.04, and thus was likely built against [this package](https://packages.ubuntu.com/bionic/libjpeg8-dev).

You will need to either find an equivalent package for Debian that deploys that major release of libjpeg, add said shared object (SO) to your `LD_LIBRARY_PATH`, use the [carla docker image](https://carla.readthedocs.io/en/latest/build_docker/), or build from source for Debian 10.5.

germanros1987, 2020-09-24:
@qhaas answer is right. We don't have plans to release for other distributions for the time being.

closed 2020-09-24 by germanros1987

thillRobot, 2020-10-17:
I am having a similar issue while trying to run the PythonAPI in a carlasim/carla:0.9.10.1 docker container. I was having a similar issue in 0.9.10 involving libxerces-c so I switched based on the changelog. 

The server runs with some warnings, but the client does not run and shows the following error when I run the client inside the container.

```
  File "PythonAPI/examples/manual_control.py", line 77, in <module>
    import carla
  File "/home/carla/PythonAPI/carla/dist/carla-0.9.10-py3.7-linux-x86_64.egg/carla/__init__.py", line 8, in <module>
  File "/home/carla/PythonAPI/carla/dist/carla-0.9.10-py3.7-linux-x86_64.egg/carla/libcarla.py", line 9, in <module>
  File "/home/carla/PythonAPI/carla/dist/carla-0.9.10-py3.7-linux-x86_64.egg/carla/libcarla.py", line 7, in __bootstrap__
ImportError: libjpeg.so.8: cannot open shared object file: No such file or directory

```

Has anyone successfully run the PythonAPI inside a carla docker container? I have read #2909, and #3236, but I have not found a solution yet. I am new to docker and carla so it may be something simple that I am doing wrong.

qhaas, 2020-10-17:
> Has anyone successfully run the PythonAPI inside a carla docker container?

Yes.  The CARLA docker container is a runtime container for the server, not the client.  You will need to start/attach a root shell to the container (e.g. for attach `docker exec -it -u 0:0 INSTANCE_ID /bin/bash`) and `apt-get install` the applicable jpeg, tiff, and png libraries.

thillRobot, 2020-10-17:
> External Email Warning
>
> This email originated from outside the university. Please use caution when opening attachments, clicking links, or responding to requests.

I appreciate the information and quick response. I have just tried your suggestion inserting the ID of my running container. In bash apt-get is unable to locate the package 'libjpeg'.

Would you prefer for me to respond directly or on github?

Thank You,
Tristan Hill

qhaas, 2020-10-17:
> In bash apt-get is unable to locate the package 'libjpeg'.

You won't; there exist several implementations of the jpeg specification in the Ubuntu repository under different package names.  In apt-based distros, you can use `apt-file search libjpeg.so.8` to identify the package that deploys the file(s) of interest.  For yum/dnf based systems, the equivalent is `yum provides`.  In the case of the needed jpeg runtime library, it is [libjpeg-turbo8](https://packages.ubuntu.com/bionic/libjpeg-turbo8)

thillRobot, 2020-10-18:
OK, I was able to install the libraries from the packages libjpeg-turbo8 and libtiff-dev using apt-get. I am now encountering an error involving the fonts when running manual_control.py or automatic_conrol.py. 

`sudo docker exec -e PYTHONPATH=/home/carla/PythonAPI/carla/dist/carla-0.9.10-py3.7-linux-x86_64.egg <container_id> python3 PythonAPI/examples/manual_control.py`

The client starts, but then it fails with this error. 

```
File "PythonAPI/examples/manual_control.py", line 507, in <listcomp>
    fonts = [x for x in pygame.font.get_fonts() if font_name in x]
TypeError: argument of type 'NoneType' is not iterable
```

Do you have any information about this error? It looks similar to #1475.

qhaas, 2020-10-19:
> OK, I was able to install the libraries from the packages `libjpeg-turbo8` and `libtiff-dev` using `apt-get`. I am now encountering an error involving the fonts when running `manual_control.py` or `automatic_conrol.py`.

I'd first verify that the CARLA PythonAPI Client is working as expected by using their bundled example 'tutorial.py', I use that as a brief 'integration test' when deploying the client.

I haven't encountered that problem in particular, but that might be because I try to deploy python packages via `apt-get` rather than with `pip3` when they are available in the Ubuntu repository.  When I do an `apt-get install python3-pygame`, I see a number of font packages being deployed alongside it.  Perhaps one of those is the one that is missing?

goksanisil23, 2021-03-11:
hi @thillRobot I have the same problem with running `pygame.font.get_fonts()` inside Ubuntu 18.04 docker, with pygame installed via pip3. However, it cannot be fixed by replacing mono with aerial since get_fonts() is returning `[None]`.

Its strange since `pygame.font.get_default_font()` returns `'freesansbold.ttf'` , so there is at least 1 available font. 

Therefore, I've just done `mono = 'freesansbold'` as a dirty fix to run it inside the docker, but would be good to know if you've figured out the root cause of this issue.

thillRobot, 2021-03-11:
Thanks for the information, @goksanisil23. I have been been compiling from source, and I have not found the cause of the issue. I will be revisiting the docker options soon, and I will try your fix.

cfanatic, 2021-06-20:
I am also running Debian, but I was unable to get CARLA running smoothly in a Docker Ubuntu container. It turned out I had to build CARLA from source. Check out https://cfanatic.medium.com/building-carla-on-debian-gnu-linux-11-68d68fddf862 if you are interested to learn how.

drmateo, 2021-10-16:
> > hi @thillRobot I have the same problem with running `pygame.font.get_fonts()` inside Ubuntu 18.04 docker, with pygame installed via pip3. However, it cannot be fixed by replacing mono with aerial since get_fonts() is returning `[None]`.
> > Its strange since `pygame.font.get_default_font()` returns `'freesansbold.ttf'` , so there is at least 1 available font.
> > Therefore, I've just done `mono = 'freesansbold'` as a dirty fix to run it inside the docker, but would be good to know if you've figured out the root cause of this issue.
> 
> I had the exact same issue, and I believe I have resolved it. The pygame.fonts module uses the CLI tool 'fc-list' to get the system fonts list. https://github.com/pygame/pygame/blob/45acd6f14555587e4fe44c5e033c0c9b9b68e2a8/src_py/sysfont.py#L232 That has to be installed in your container by `apt install fontconfig`

same problem and, yes, I resolved installing fontconfig

thillRobot, 2021-10-16:
Thanks for the information @mkhansenbot and @drmateo. I will give this a try.

ElectricRCAircraftGuy, 2022-11-04:
> To install `libjpeg8-dev`, use this:
> 
> ```bash
> sudo apt install aptitude
> sudo aptitude install libjpeg-dev
> ```
> 
> You may then need to choose "no", "yes", "yes". See my detailed instructions here: [How to install/upgrade to the latest version of Poppler/`pdftoppm` (version 22.11.0 at the time of this writing) on Linux Ubuntu](https://stackoverflow.com/a/74313286/4561887)

See my answer: https://unix.stackexchange.com/a/723619/114401