carla-simulator/carla #3237: Can I call Actor.Destroy() safely in LocalizationStage.cpp?

youngsend, 2020-09-01:
Hello,
For some reasons, my map's roads are not closed, which means vehicles may fall into abyss from the map.
To account for this, I want to destroy the vehicle in LocalizationStage.cpp when the vehicle is going to fall.

I know commands are executed in BatchControlStage.cpp.
But I also find that in CollisionStage, TrafficLightStage, MotionPlannerStage and BatchControlStage, every vehicle's `IsAlive()` is checked.
So I think it's OK to call `Destroy()` safely in LocalizationStage.cpp.
Am I right?

Do you have any suggestions?
Besides, when calling `Destroy()`, do I need to consider whether it's synchronous mode or asynchronous mode?

youngsend, 2020-09-01:
I found the `LocalizationStage::TryDestroyVehicle(const Actor& actor)`.
Maybe I can just use it.

jackbart94, 2020-09-01:
Hi,

That ``TryDestroyVehicle`` is a function used for vehicle destruction in case some of them get stuck and don't move for a certain amount of time. I wouldn't recommend using it directly, but you can analyze it as an example of how to destroy a vehicle within Traffic Manager and make new functions for your specific case.

In your case, you might just insert a check if a vehicle is falling down (pretty sure that's easy to achieve) and delete it in the ALSM.cpp class.

There are many ways to do this, you should even be able to do it inside the Python script (although modifying the API would look cleaner).

youngsend, 2020-09-02:
Thank you! @jackbart94 
Sorry that I forgot to say that I am using carla 0.9.8 and there's no ALSM.cpp class in it.
Actually when one vehicle is going to fall down, exception happens at the following code because `next_waypoints` is empty (in the `// Populating the buffer` part):
```c++
std::vector<SimpleWaypointPtr> next_waypoints = waypoint_buffer.back()->GetNextWaypoint();
uint64_t selection_index = 0u;
// Pseudo-randomized path selection if found more than one choice.
if (next_waypoints.size() > 1) {
   selection_index = static_cast<uint64_t>(rand()) % next_waypoints.size();
}
SimpleWaypointPtr next_wp = next_waypoints.at(selection_index);
```
Since there's no ALSM.cpp class in carla 0.9.8, could you tell me where is the best place to delete a vehicle?
And since exception happens in `LocalizationStage.cpp`, maybe I should just check and delete a vehicle here.

jackbart94, 2020-09-02:
Hi,

Yes, in 0.9.8 it's probably more appropriate to delete them inside the Localization Stage. You can probably re-use the code from there. The ``max_idle_time.first`` is the maximum a vehicle has to be stopped before the elimination policy destroys it. There is another parameter which basically tells that you can only destroy a vehicle each X seconds, can't remember the name for 0.9.8.

What you should do anyway is just make your own policy and destroy vehicles like in the vehicle destruction functions, just without any time constraint.. I'd rather use either the fact that there are no more waypoints in the list, or the z value (since vehicles are dropping down).

youngsend, 2020-09-03:
Thank you @jackbart94 !
I re-used code from `TryDestroyVehicle` without the time constraint part. It seems that `python3 spawn_npc.py` does not abort any more. I am going to take a video on Unreal Engine to show the vehicle destruction.

closed 2020-09-03 by youngsend

AOOOOOA, 2023-08-30:
> Hi,
>
> That `TryDestroyVehicle` is a function used for vehicle destruction in case some of them get stuck and don't move for a certain amount of time. I wouldn't recommend using it directly, but you [...]

Hi, I'm also curious about why the threshold is 10 seconds. Do you have any basis for that?