carla-simulator/carla #3238: How run two server and two client

XieJinLiang, 2020-09-01:
Hi,
Now I want to run two servers and two corresponding clients. I ran the following two terminal commands in my carla directory:
`./CarlaUE4.sh`
`./CarlaUE4.sh --world-port=20000 -carla-streaming-port=8001`

then I run `python manual_control.py -p 2000` and `python manual_control.py -p 20000` to generat two clients.
Why the second generated client cannot use the autopilot function. I can’t even use the arrow keys to control the second vehicle.Do I need to add other parameter settings?

In addition, if I start two servers through different ports, can I get the information of the first server through only one client, then disconnect, and then connect to the second server to get data?

jackbart94, 2020-09-24:
Hi,

I'll try to answer this, the issue is that you need to also create the Traffic Manager on two different ports. So somewhere in the ``manual_control.py`` script, probably in the ``game_loop``, you need to do ``get_trafficmanager`` and have an argument to set the port, then create two different TMs when launching the script.

Here is the documentation about [Multi-Simulation and Multi-TM](https://carla.readthedocs.io/en/latest/adv_traffic_manager/#multitm).

Let me know if this fixes it!

XieJinLiang, 2020-09-25:
Hi, @jackbart94 
Thanks for you reply.It work!
I run two carla server with two different `-carla-rpc-port`. Here I add the argument `--tm-port `in the `game_loop,` and then I run the `manual_control.py` script twice with different ` -p` and `--tm-port`. Now I can run two clients and and let them autopilot at the same time, thank you very much

closed 2020-12-25 by stale[bot]