carla-simulator/carla #3239: Carlar Simlator：The principle of IMU

BoldArro1, 2020-09-01:
Hi,
I've recently been learning the source code of IMU in Carla.Some formulas are used in the process of calculating the results of accelerometer, gyroscope, and compass outputs.Without understanding the principles of IMU simulation, just learning the code made me feel very confused。
So, can you provide me with the papers or books that you refer to when you design IMU simulations?
Thank you！

GabiShahmayster, 2020-09-01:
@BoldArro1 
Hi.
Are you looking for information regarding an IMU principal of operation or the IMU errors model?
Gabi

BoldArro1, 2020-09-02:
> 
> 
> @BoldArro1
> Hi.
> Are you looking for information regarding an IMU principal of operation or the IMU errors model?
> Gabi
YES
I found some descriptions in the code file, but based on those descriptions, I didn't find the right mathematical model.

GabiShahmayster, 2020-09-02:
@BoldArro1 
Hi,
I haven't looked at the C++ source, but based on their parameters and some basic testing, they appear to be using the following model:
gyro_i[k] = gyro_true_i[k] + noise_gyro_bias_i + gyro_wideband_noise_i[k]
where:
gyro_i[k] - gyro measurement, at k-th time sample
gyro_true_i[k] - true rotation-rate, at k-th time sample, projected on IMU's i-th axis
noise_gyro_bias_i - constant bias on IMU's i-th axis
gyro_wideband_noise_i[k] - wideband noise ("arw", Angular Random Walk) term, at k-th time sample, drawn from N(0,noise_gyro_stddev_i^2)

the accelerometer should follow a similar model, without a constant bias term
accelerometer_i[k] = accelerometer_true_i[k] + accelerometer_wideband_noise_i[k]
where:
accelerometer_i[k] - accelerometer measurement, at k-th time sample
accelerometer_true_i[k] - true acceleration, at k-th time sample, projected on IMU's i-th axis
accelerometer_wideband_noise_i[k] - wideband noise ("vrw", Velocity Random Walk) term, at k-th time sample, drawn from N(0,noise_accel_stddev_i^2)

as a side-note, you can see that the IMU error model, per axis, is basically drawing from a N(noise_gyro_bias_i, noise_gyro_stddev_i^2) for the gyro and from N(0,noise_accel_stddev_i^2) for the accelerometer, but in classical IMU error modeling, the biases are separated from the wideband noise term (and they follow a more realistic model, for low-cost sensors...)

Hope that helps,
Gabi Shahmayster

BoldArro1, 2020-09-03:
Yes,It is helpful.
However,there are still some questions that bother me.According to your description，the result of imu is composed of truth ，bias，and noise.
In the code, the conversion from the actor coordinate system to the world coordinate system is used. If I take the data directly from the world coordinate system, can I get the result in the world coordinate system plus the noise to get the IMU simulation result?

GabiShahmayster, 2020-09-03:
All IMU errors should be simulated and added in the actor frame.
Please note that there was a bug in IMU gyroscopes simulation, which should have been resolved in version 0.9.9.10 (https://github.com/carla-simulator/carla/pull/2998)

BoldArro1, 2020-09-03:
Please tell me what is the origin of the Actor scene as coordinates. I know that Actor is a scene defined in ue4, but I can't find a description of the coordinate axis.
Thank you

GabiShahmayster, 2020-09-03:
Are you asking about Carla's coordinate system?
It's built using Unreal Engine and uses it's coordinate system. 
It's a left-handed Cartesian reference frame. Each actor has a Location (3d translation) and Rotation (3 Euler angles) attribute.
Perhaps this is too simplistic and I didn't understand your question... Carla's documentation is very good, you should a lot of information there...

BoldArro1, 2020-09-04:
I think I've found the answer.
The carla's actor scene uses the left hand coordinate system, the opendrive format map uses the right hand coordinate system, and my simulation in the opendrive map needs to convert the right hand coordinate system to the left hand coordinate system, and then to the world coordinate system. I can get the simulation result according to carlad's calculation method.
Thank you！

closed 2020-09-24 by germanros1987

GabiShahmayster, 2021-10-15:
@werewolfdev 

Please see issue #2737.

Good luck!
Gabi

GabiShahmayster, 2021-10-16:
@werewolfdev 
As far as I understood, they were looking for a way to convert the outputs (geographic coordinates) of the simulated GNSS sensor, which can be added to the Carla ego-vehicle (just like a camera, radar and lidar), to the "native" Unreal engine coordinates, which are reported by Carla, for every actor/sensor.

werewolfdev, 2021-10-17:
I think the georeferencing of aCARLA Map is described in the GeoReference Tag inside the xodr file . I could find lat lon and alt as 0,0,0 inside the Town.xodr. Now this means if Iam using a real world coordinate system like ECEF ,then both coordinate system origin is at 0,0,0 . But still since the CARLA frame is in East South Up ,Iam not sure on how to find the rotation matrix involved

werewolfdev, 2021-10-17:
@GabiShahmayster maybe I am missing out on core concepts on how to implement this .(ENU coordinates from real car)<-> Coordinates in CARLA. I will give a detailed study of #2737 and see .Thanks

werewolfdev, 2021-10-17:
Hi @GabiShahmayster so i looked into the issue and understood how a geodetic point obtained from the GNSS sensor attached to the ego vehicle in CARLA world can be converted into CARLA x,y,z coordinates (East South Up). What I would like to perform is to convert the x,y,z coordinates obtained from CARLA to  the coordinate frame of the real world (ENU) and send it back to the original CAR ECUs to test the different driver assistant functions . 

So can you convey your thoughts on this : since the geo origin in the real world and carla map (Town 01) is same . 
How to find the trasnformation matrix between the
the ENU coordinates from the real world x,y,z and yaw,pitch, roll
the ENU coordinates from the CARLA after inverting the y coordinate . 
Since the ENU is a tangential plane system and ties to a particular geodetic coordinates ,Iam confused on how to move forth

GabiShahmayster, 2021-10-17:
@werewolfdev 
Sorry but it's a little hard to understand your question.
Let's start with something basic.
As far as I know, most AD functions are related to, or driven by, the vehicle's perception sensors (camera, lidar, radar, ultrasonic, etc), hence no geographic positioning information is required. The standard practice in such a scenario would be to use a common reference-frame (such as the vehicle longitudinal-lateral-vertical axes) for the perception (object detection, lane detection, etc) and the control (longitudinal-lateral control).
On the other hand, it's possible that you trying to investigate something related to gnss-based lane keep or perhaps something related to HD maps. In that case, I can understand your need for geographic coordinates.
Which of these are you after?

werewolfdev, 2021-10-18:
Hey sorry @GabiShahmayster  I will try to explain a bit. So lets consider my project consists of two modules . One module is the original car on the test track which is sending out its position data in ENU coordinates. (x,y,z,yaw,pitch,roll) as input data to CARLA to generate the transformation matrix between the two coordinate systems. Now the second module is CARLA Simulator. The output I need from the CARLA to the external vehicle ECU is the position of the actors in the coordinate system in which we received from the vehicle . The basic idea is I will be using this CARLA output data for bypassing the original sensor data of the vehicle on the test ground and test different AD functions like Emergency breaking , Turn assist , Evasive Steering Assists and so on. 
So basically at every frame I would like to use the transformation matrix to convert the actor positions from CARLA back to the original coordinates on the test ground . I hope you could get an idea now . Please let me know otherwise. Thanks a lot in advance.

werewolfdev, 2021-10-18:
SO now we have coordinates in ENU frame on both sides. But both could be on different local tangential planes right

werewolfdev, 2021-10-18:
Yeah surely . So is it like if I have index 1 ENU frame coordinates from the car side and Index 2 ENU coordinates from the CARLA side . So a pair of coordinates in the same frame and I just have to find how to transform between these two if Iam not wrong. A translation and a rotation matrix 

Ok thanks for the point on INS.

werewolfdev, 2021-10-18:
And just one more thing , I should be using the ROS bridge for performing all this and interact with the external systems right

werewolfdev, 2021-10-18:
ok np. and  @GabiShahmayster i I thought about it and I think what you mentioned here is right :**Is this correct? Are you looking for the transformation between ENU and Vehicle frames?** Could you pls give me basic idea on how to move ´forward with this transformation

GabiShahmayster, 2021-10-18:
You're probably going to have to catch up on transformations between coordinate frames.
Try robotics textbooks or online resources, they're great for learning about homogeneous transformation matrices. 
Good luck!

TzabarDolev, 2023-09-18:
I wrote a short article about IMUs and they're usage in CARLA in my research blog - https://carlasimblog.wordpress.com/2023/09/16/understanding-imu-sensors-and-their-role-in-autonomous-driving-simulations-with-carla/.
Hope you find it helpful.