carla-simulator/carla #3240: "clang 8 is required, but it's not installed"

atsushinakazima, 2020-09-01:
I followed this site.　https://carla.readthedocs.io/en/0.9.6/how_to_build_on_linux/
However, I got the following message in the "make launch" section of Build CARLA.

"clang 8 is required, but it's not installed.
Util/BuildTools/Linux.mk:143: recipe for target 'setup' failed
make: ***[setup] Error 1"

Ubuntu version 18.04.5 LTS
Unreal Engine 4.22

However i can't install clang 8 because i don't know ubuntu well

please help what can i do?

qhaas, 2020-09-04:
> However i can't install clang 8 because i don't know ubuntu well

Hopefully this doesn't come across as too blunt, because it isn't my intentions to be rude...  One likely shouldn't go down the path of compiling CARLA from source without a working knowledge of Debian-like linux distributions (e.g. Ubuntu) and sudo/root access to deploy dependencies.  Attempting to use a different version of clang (or UE for that matter) than the one supported by the version of CARLA you are using will lead you into problems like in Issue #3246 , which are much worse than the learning curve of Ubuntu's package management system `apt`.

There are [vendor supplied binaries](https://carla.org/2019/07/12/release-0.9.6/) and a [docker container](https://carla.readthedocs.io/en/latest/build_docker/) if your Linux environment doesn't support said binaries (e.g. glibc version mismatches) that will allow you to avoid compiling from source.

Also, [the version of clang supported by carla 0.9.6 is clang 7](https://github.com/carla-simulator/carla/blob/0.9.6/Util/BuildTools/Setup.sh#L8), not clang 8.  The documentation linked is for CARLA 0.9.6.  Are you using the right tag/release of the CARLA source and/or the right documentation's version?  If you are using CARLA 0.9.9.x, the build instructions are [here](https://github.com/carla-simulator/carla/blob/0.9.9.4/Docs/build_linux.md) and show you how to deploy clang-8 in Ubuntu 18.04.

atsushinakazima, 2020-09-05:
I really need carla0.9.6.
I currently have nvidia440 and cuda10.1.243 in my system. Because this is within the recommendation of carla0.9.6, I do not think that there is not a problem especially...(By the way, the reason why nvidia is 440 is nvidia, I thought to specify 418 at first and install it, but because my PC does not boot Ubuntu like this site
https://itectec.com/ubuntu/ubuntu-problems-with-booting-the-system/
, I use 440  by $sudo ubuntu-drivers autoinstall   )

I think the Ubuntu kernel is fine too, but I still get the same clang error as before.
By the way, unreal engine 4.22 starts with no problem, and during the build of carla. It goes to /Update.sh with no problem.
I describe the result of nvidia-smi as follows. Is there any solution to build carla 0.9.6? Please help me out.

atusi@atusi-G3-3500:~/carla$ uname -rsv
Linux 5.4.0-42-generic #46~18.04.1-Ubuntu SMP Fri Jul 10 07:21:24 UTC 2020
atusi@atusi-G3-3500:~/carla$ nvidia-smi
Sat Sep  5 17:40:49 2020       
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 440.100      Driver Version: 440.100      CUDA Version: 10.2     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|===============================+======================+======================|
|   0  GeForce GTX 166...  Off  | 00000000:01:00.0 Off |                  N/A |
| N/A   52C    P8     5W /  N/A |    243MiB /  5944MiB |      4%      Default |
+-------------------------------+----------------------+----------------------+
                                                                               
+-----------------------------------------------------------------------------+
| Processes:                                                       GPU Memory |
|  GPU       PID   Type   Process name                             Usage      |
|=============================================================================|
|    0      1425      G   /usr/lib/xorg/Xorg                           108MiB |
|    0      1599      G   /usr/bin/gnome-shell                         131MiB |
+-----------------------------------------------------------------------------+

atsushinakazima, 2020-09-05:
https://mail.google.com/mail/u/1?ui=2&ik=7fcc80d93b&attid=0.1&permmsgid=msg-a:r5938779272112668039&view=att&disp=safe&realattid=f_keph73060

atsushinakazima, 2020-09-05:
[image]

atsushinakazima, 2020-09-06:
It looked like I had an $sudo apt update between
$sudo apt-get install build-essential clang-7 lld-7 g++-7 cmake ninja-build libvulkan1 python python-pip python-dev python3-dev python3-pip libpng16-dev libtiff5-dev libjpeg-dev tzdata sed curl unzip autoconf libtool rsync

and 

$pip2 install --user setuptools

, is it possible that this is causing the above error?

[image]

germanros1987, 2020-09-24:
+@bernatx I know it's a busy time, but could you provide some help?

bernatx, 2020-09-24:
As **qhaas** sais, are you using the right tag/release of the CARLA source and/or the right documentation version?
It seems that you are trying to compile a latest version of CARLA (which needs clang 8) instead of the CARLA 0.9.6 that you need. Please, download the source for the TAG 0.9.6
The message about **clang 8 is required** only shows when compiling for CARLA 0.9.9.x and UE 4.24

closed 2020-12-25 by stale[bot]

mullerhai, 2021-12-21:
HI ,if you has install llvm@8 [clang-8] ,maybe the install location is not the same the make launch  script : carla/Util/BuildTools/Setup.sh,you need to change the script the clang-8 location as you specify .
example :

 ##原来的
 
 command -v /usr/bin/clang++-8 >/dev/null 2>&1 || {
  echo >&2 "clang 8 is required, but it's not installed.";
  exit 1;
}

CXX_TAG=c8
export CC=/usr/bin/clang-8
export CXX=/usr/bin/clang++-8
 ```
 
 ```
 #以实际位置为准
 command -v /usr/local/opt/llvm@8/bin/clang++ >/dev/null 2>&1 || {
  echo >&2 "clang 8 is required, but it's not installed.";
  exit 1;
}

CXX_TAG=c8
export CC=/usr/local/opt/llvm@8/bin/clang-8
export CXX=/usr/local/opt/llvm@8/bin/clang++
 
 ```

hemeda3, 2023-01-06:
I had to do `sudo apt-get install clang` in ubuntu 20.04 to make it work