carla-simulator/carla #3245: make launch, Windows 0.9.x error

Inezzzz, 2020-09-01:
I'm building on Windows and keeps getting an error on boost. I've tried to remove boost folder completely from carla/Build and re-run "make launch" or "make LibCarla" multiple times, still the same error. I also tried to install boost directly into carla/Build which seemed to be successful but then I got same error after try "make launch" again.  All commands were in x64 command window.
[image]

H-Whisky, 2020-09-02:
You should go to "~/carla/Util/InstallersWin", then you can see install_boost.bat, rending the content of the file shows that you can go to https://dl.bintray.com/boostorg/release/ to download Specify version *.zip file, finally put the *.zip into "~/carla/Build", and run "make launch".

Inezzzz, 2020-09-02:
> You should go to "~/carla/Util/InstallersWin", then you can see install_boost.bat, rending the content of the file shows that you can go to https://dl.bintray.com/boostorg/release/ to download Specify version *.zip file, finally put the *.zip into "~/carla/Build", and run "make launch".
Tried to render the install_boost.bat and this is what I got

C:\carla\Util\InstallersWin>install_boost
    -[install_boost]: [Batch params]:
    -[install_boost]: You must specify a boost version using [-v|--version]
    -[install_boost]: Exiting with error...

How do I specify the boost version? I downloaded a boost binary and put in /carla/Build

H-Whisky, 2020-09-02:
> C:\carla\Util\InstallersWin>install_boost
> -[install_boost]: [Batch params]:
> -[install_boost]: You must specify a boost version using [-v|--version]
> -[install_boost]: Exiting with error...
>
> How do I [...]

Go to https://dl.bintray.com/boostorg/release/ , find and download *0.72.0*.zip, put it into ~/carla/Build/ when download finished.

germanros1987, 2020-09-24:
@Inezzzz have you solved your problem? Otherwise, @marcgpuig could you please take a look?

Inezzzz, 2020-09-24:
Nope. Still not working, kinda gave up on CARLA, but would love to try
again if the team’s going to take a look.

nagasanjay, 2020-10-10:
Have you installed `c++ workload and .NET workload` from visual studio installer ?

closed 2020-12-25 by stale[bot]