carla-simulator/carla #3246: Have Issue with Make PythonAPI with Carla version 0.9.9.4

KasraMokhtari2112, 2020-09-02:
Hi Carla team,

I am trying to build Carla (0.9.9.4) from source following this link: https://carla.readthedocs.io/en/latest/build_linux/ and I have an issue with "make PythonAPI", although "make launch", "make LibCarla" work for me. The issue is "make PythonAPI.2" works fine which is for python2 but for python 3 this command is not working. This is the error I am getting:

BuildPythonAPI.sh: Building Python API for Python 3.
compiling:
  - source/libcarla/libcarla.cpp
running bdist_egg
running egg_info
writing manifest file 'source/carla.egg-info/SOURCES.txt'
running install_lib
running build_py
running build_ext
clang-8 -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -g -flto -fuse-linker-plugin -ffat-lto-objects -fPIC -Idependencies/include -I/usr/include/python3.6m -c source/libcarla/libcarla.cpp -o build/temp.linux-x86_64-3.6/source/libcarla/libcarla.o -isystem dependencies/include/system -fPIC -std=c++14 -Werror -Wall -Wextra -Wpedantic -Wno-self-assign-overloaded -Wdeprecated -Wno-shadow -Wuninitialized -Wunreachable-code -Wpessimizing-move -Wold-style-cast -Wnull-dereference -Wduplicate-enum -Wnon-virtual-dtor -Wheader-hygiene -Wconversion -Wfloat-overflow-conversion -DBOOST_ERROR_CODE_HEADER_ONLY -DLIBCARLA_WITH_PYTHON_SUPPORT -DLIBCARLA_IMAGE_WITH_PNG_SUPPORT=true
clang: error: optimization flag '-fuse-linker-plugin' is not supported [-Werror,-Wignored-optimization-argument]
clang: error: optimization flag '-ffat-lto-objects' is not supported [-Werror,-Wignored-optimization-argument]
error: command '/usr/bin/clang-8' failed with exit code 1
Util/BuildTools/Linux.mk:87: recipe for target 'PythonAPI' failed
make: *** [PythonAPI] Error 1

--------------------------------------------------------
gcc (Ubuntu 7.5.0-3ubuntu1~18.04) 7.5.0
clang version 8.0.1- (branches/release_80)

Any idea would be appreciated? Thanks for your time!

olitazl, 2020-09-03:
Same problem here while building it within the docker container.
Help would be really nice!

tia

baurst, 2020-09-03:
I had the same issue building in Docker. 

1. It seems that -Werror picks up the warning about unused optimization flags, thus I removed -Werror from PythonAPI setup.py, around line 59. This fixed the issue if I remember correctly, but then the next problem occured in the line after that in setup.py.
"build/temp.linux-x86_64-3.6/source/libcarla/libcarla.o: file not recognized: File format not recognized"
This error seemed really odd, so I compared setuptools versions and outputs of docker vs ubuntu 18.04 that I am working on...
2. Docker was using setuptools >= 52 (don't remember the exact version), so a downgrade to setuptools==45.0 fixed the second issue as well. Now it builds again.

I realize both "fixes" are not really a fix but a hack, but maybe this hotfix can help you to continue with your work until there is a proper fix. Good luck!

qhaas, 2020-09-04:
> 1. It seems that -Werror picks up the warning about unused optimization flags, thus I removed -Werror from PythonAPI setup.py, around line 59. This fixed the issue if I remember correctly, but then the next problem occured in the line after that in setup.py.

I encounter these errors when using a (major) version of clang that isn't the officially supported version of clang with the version of carla I am using.  Clang typically 'improves' its abilities to [find possibly problematic code at compile time](https://en.wikipedia.org/wiki/Static_program_analysis) with each major release.  The result is [new types of warnings](https://clang.llvm.org/docs/DiagnosticsReference.html).  When `-Werror` is specified, that causes clang to treat warnings as errors and halt compilation.  Warnings are not errors, but they could be smoke indicating a future fire.

Ideally, one would use the same version of clang used by the CARLA developers (e.g. CARLA 0.9.7 that is clang-7, CARLA 0.9.9.x that is clang-8, etc.), the CARLA developers have already identified the 'harmless' warnings and addressed/ignored them with the supported clang release.

Otherwise, as baurst stated, you are looking at dropping the `-Werror` flag, which is likely not as bad as it sounds, especially if you are not modifying the CARLA code.  You can still see the warnings in the build logs, either that or find the warning that is causing the failure and prepend `-Wno-` to the class of warning you are seeing to the code location in setup.py that baurst indicated (e.g. `'-Wno-defaulted-function-deleted'`).

sagar-g-v, 2020-09-10:
I got the same error. now it is fixed

try running below command before  `make PythonAPI`
```bash
pip3 install --user setuptools 
pip2 install --user setuptools
```

if it fails try this
```bash
pip3 install setuptools 
pip2 install setuptools
```

closed 2020-09-24 by germanros1987