carla-simulator/carla #3247: Import RoadRunner Map in CARLA/ Build CARLA Ubuntu 16.04

souvikboseGM, 2020-09-02:
I am  trying to build CARLA in Ubuntu 16.04  
and using this document  https://carla.readthedocs.io/en/stable/how_to_build_on_linux/#build-unreal-engine

Unreal Engine build is done. 
and Then I used "git clone https://github.com/carla-simulator/carla" to download CARLA 0.9.8 [No particular reason to download 0.9.8 . My main aim is to Import RoadRunner Map in CARLA]

Then While doing " ./Setup.sh ". I got below error 
[image]

The "https://carla.readthedocs.io/en/stable/how_to_build_on_linux/#build-unreal-engine" didnot asked for clang 7 ? .

Can you help me to solve this ?

pergustav, 2020-09-02:
You are reading the docs for the stable version, which isn't 0.9.8.
You will probably have better luck with this one. https://carla.readthedocs.io/en/latest/build_linux/

souvikboseGM, 2020-09-02:
@pergustav Thanks for the docs. I just moved to this doc. I am facing now new issue. Posted with Error No #3248 . If you could pls  check ,if you could help

souvikboseGM, 2020-09-04:
Solved by https://carla.readthedocs.io/en/0.9.8/build_linux/

closed 2020-09-04 by souvikboseGM