carla-simulator/carla #3252: Tracking of  #3251

marcgpuig, 2020-09-02:
_This issue is intended to track #3251._

It must be tested in Windows (@bernatx and @XGodina) and Linux (@Axel1092).
- Make sure the client colors mimic the original UE4 colors.
- Make sure all the semantic tags are correctly displayed.

Axel1092, 2020-09-03:
Works!

bernatx, 2020-09-03:
Works for me also, in windows.

closed 2020-09-15 by marcgpuig

pangxiao2013, 2020-09-25:
> _This issue is intended to track #3251._
> 
> It must be tested in Windows (@bernatx and @XGodina) and Linux (@Axel1092).
> 
> * Make sure the client colors mimic the original UE4 colors.
> * Make sure all the semantic tags are correctly displayed.

> Works for me also, in windows.

Where can we download 0.9.10?

XGodina, 2020-09-29:
Hi @pangxiao2013 

You can download here: http://carla.org/2020/09/25/release-0.9.10/