carla-simulator/carla #3255: Physics precision: lower fixed_delta_seconds does not necessarily improve precision

hh0rva1h, 2020-09-02:
**Carla Version:** 0.9.9.4, binary release tarballs
**Environment:** Ubuntu 20.04

### Description

We tried to find an optimal (tradeoff between precision and performance) fixed_delta_seconds value for our reinforcement learning environments. We noticed however that a lower value does not always make the physics more sane, sometimes accelerations start to fluctuate heavily.

Consider the following code, which sets up a vehicle in a scene and makes it accelerate and brake, afterwards plots velocities and accelerations.

```
import carla
from math import sqrt

# initialise carla with our default map
client = carla.Client("localhost", 2000)
client.set_timeout(5)
world = client.load_world("Town06")

settings = world.get_settings()
settings.fixed_delta_seconds = 0.045
settings.synchronous_mode = True
settings.no_rendering_mode = True
world.apply_settings(settings)
blueprints = world.get_blueprint_library()

print(world.get_map().get_spawn_points()[1])

# spawn our favourite car
default_car = blueprints.find("vehicle.tesla.model3")
#position = carla.Transform(carla.Location(x=350, y=12.5, z=3), carla.Rotation(yaw=180))
position = carla.Transform(carla.Location(x=626.831238, y=-13.563108, z=0.300000), carla.Rotation(yaw=180))
vehicle = world.spawn_actor(default_car, position)

for i in range(20):
    try:
        world.tick()
    except:
        print("Skipped tick")

weather = carla.WeatherParameters(
    cloudiness=80.0,
    precipitation=30.0,
    sun_altitude_angle=70.0)

world.set_weather(weather)

# workaround unrealistic Carla phenomenas
physics_control = vehicle.get_physics_control()
physics_control.gear_switch_time=0.01
physics_control.damping_rate_zero_throttle_clutch_engaged=physics_control.damping_rate_zero_throttle_clutch_disengaged
vehicle.apply_physics_control(physics_control)
vehicle.apply_control(carla.VehicleControl(throttle=0, brake=1, manual_gear_shift=True, gear=1))
for i in range(9):
    try:
        world.tick()
    except:
        print("Skipped tick")

world.get_spectator().set_transform(position)

#exit()

# full throttle in the beginning
vehicle.apply_control(carla.VehicleControl(throttle=1.0))
print("Full throttle")
for _ in range(20):
    try:
        world.tick()
    except:
        print("Skipped tick")

# use norm of velocity and accel vectors (maybe try to disregard z?)
def norm(vector):
    return sqrt(vector.x**2 + vector.y**2 + vector.z**2)

# log all vehicle data in lists
accel_log = []
velocity_log = []
throttle_log = []
brake_log = []
z_log = []

# accelerate up to 20m/s and then apply throttle=0.82 to stay there until set location
throttle_changed = False
throttle = 1.0
while vehicle.get_location().x > 0:
    try:
        world.tick()
    except:
        print("Skipped tick")
    a = vehicle.get_acceleration()
    a_x = 0 if a.x == 0 else -norm(a)*a.x/abs(a.x)

    v = vehicle.get_velocity()
    v_x = 0 if v.x == 0 else -norm(v)*v.x/abs(v.x)
    if not throttle_changed and v_x > 20: # 20 40
        throttle = 0.475 # 0.48 0.77
        print(f"Apply throttle={throttle} and hold speed")
        throttle_changed = True

    vehicle.apply_control(carla.VehicleControl(throttle=throttle))

    accel_log.append(a_x)
    print(a_x)
    velocity_log.append(v_x)
    throttle_log.append(throttle)
    brake_log.append(0.0)
    z_log.append(a.z)

# now change throttle and brake to see the impact on the velocity
throttle = 0.0
brake = 0.0

print(f"Apply throttle={throttle}, brake={brake}")
vehicle.apply_control(carla.VehicleControl(throttle=throttle, brake=brake))

# break the loop at some coordinate or when the velocity is low enough
while vehicle.get_location().x > -300:
    try:
        world.tick()
    except:
        print("Skipped tick")
    a = vehicle.get_acceleration()
    a_x = 0 if a.x == 0 else -norm(a)*a.x/abs(a.x)

    v = vehicle.get_velocity()
    v_x = 0 if v.x == 0 else -norm(v)*v.x/abs(v.x)
    if v_x < 2:
        print("Vehicle stopped")
        break

    accel_log.append(a_x)
    throttle_log.append(throttle)
    brake_log.append(brake)
    velocity_log.append(v_x)
    z_log.append(a.z)
else:
    print("Vehicle has driven outside the test area")

import matplotlib.pyplot as plt

# first plot the velocities, accels, throttle and brake values by tick number (implicit indexing)
plt.plot(velocity_log, label="v")
plt.plot(accel_log, label="a")
plt.plot(z_log, label="z cooridnate")
# throttle and brake are scaled *10 to make them readable
plt.plot([10*t for t in throttle_log], label="throttle*10")
plt.plot([10*b for b in brake_log], label="brake*10")
plt.legend()

plt.show()
```

For `settings.fixed_delta_seconds = 0.05` the plots looks as follows:
[image]
In detail:
[image]

For `settings.fixed_delta_seconds = 0.045` the plots looks as follows:
[image]
In detail:
[image]

This looks clearly worse compared to the higher value. Also it is really bad compared to `settings.fixed_delta_seconds = 0.02`:
[image]
In detail:
[image]

### Discussion

The fluctuations of the acceleration seems to be related to a tiny variation of z-coordinate of the map, however without a really small delta_second value this affects the reported accelerations values significantly which is really bad for our purposes. Even with 
```
def norm(vector):
    return sqrt(vector.x**2 + vector.y**2)
```
i.e. leaving out the z-coordinate the results are similarly bad, thus the fluctuations are not due to the z-component of the acceleration! 

Could you look into this, i.e. making the simulation more realistic even with bigger values for delta_second? Especially the oscillations happening really worry us. This is just a simple case which we tracked down, there might be many more unreliable situations in complex scenarios. We heavily rely on acceleration values and this affects as negatively since we have to make the delta_seconds value really low in order to reduce such phenomena. It's especially worring to us that such a tiny z-coordinate fluctuation can have such a great effect.

germanros1987, 2020-09-24:
I expect a productive discussion on this topic. @DSantosO please, when your plate is not too full, have a look at this.

hh0rva1h, 2020-12-08:
Still an issue.

germanros1987, 2021-01-26:
@DSantosO please, let's debug this

DSantosO, 2021-01-27:
Hello @hh0rva1h,
Sorry about the delay.

Here we need to consider several points.
We need to consider that the acceleration that we provide from the API is not the 'real' acceleration because the physics solvers usually do not provide this variable. We compute it using a simple backward finite differences derivative using the previous velocity and the current one and dividing it by the delta time. The oscillations that you see are a usual problem when computing numerical derivatives and need to be addressed carefully.
Anyway, these oscillations should not be interpreted in any way as a problem with the physical simulation, only of the way of computing the acceleration. To check this, we should look at actual physical quantities like the velocity and the position, but I will come back later to that.
In your specific case, if you want to compute the acceleration in a more accurate way, you should do it as a post-process using the velocity and/or the position. You can take advantage that you are using a fixed delta time and use a more advanced numerical differentiation method. You can start checking central finite differences from the velocity that uses also two points but instead of the current, uses the next one and should be more stable. Also you have can try a second-order derivative from the position. [Here](https://en.wikipedia.org/wiki/Finite_difference) you can check the simplest formulas. If those ones are not good enough you can try higher-order methods that involve more points but should give you better results.

The second important point is to understand what it means to decrease the delta time.
In CARLA we have two different types of delta time: the general one between render frames and the physics one. This is inherited from UE4 and PhysX and makes sense because if you want to use a dt of, let's say, 0.05s, the physical simulation would not make sense at all. For this reason, there are some physics substeps to compensate for that. In previous releases, the target physical dt was 0.016666s and now, since 0.9.11, it is by default 0.01s and can be configured via the API. [Here](https://carla.readthedocs.io/en/latest/python_api/#carla.WorldSettings) and [here](https://github.com/carla-simulator/carla/issues/3550) you can check more about that. In this context, it is understandable that the oscillations were growing with the decrease of it. The physical dt on all your simulations were the same but the formula for computing the acceleration was worse). 

I have prepared a couple of plots to show this based on a modified version of your script. They are not really convergence plots because I will need to synchronize the simulation a bit better but for matching everything but they are ok to have a broad picture.

---------------------------------------
[image]
This a plot of the velocity for several different dt.

---------------------------------------
[image]
For a constant dt=0.04 variating the physical dt with 1,2,4,8 physical substeps. Here you can see the errors that showed up when the physical dt is too high.

[image]
For the z-acceleration, it is a bit similar. With a constant dt (the one that we use to compute the acceleration), the same error as the velocity is there.

DSantosO, 2021-01-27:
By the way, I forgot to mention that in the IMU sensor, we are providing a better approximation for the acceleration(using a second-order computation from the position of the sensor). It can also worth a try to check if this works for you, but any acceleration computed using the central differences formula should be better than any backward that we could provide in runtime.

closed 2021-05-26 by DSantosO

AOOOOOA, 2024-08-22:
> Hello @hh0rva1h, Sorry about the delay.
>
> Here we need to consider several points. We need to consider that the acceleration that we provide from the API is not the 'real' acceleration because the [...]

I'm curious about the relationship between frame time and physics sub-step time. If the frame time is 0.05 and the physics sub-step time is 0.01, will the physics be processed 5 times per frame? If the frame time is 0.2, will the physics stop processing after reaching the maximum sub-steps (10 in this case)?