carla-simulator/carla #3257: How replace existed scene content with my own fbx and opendrive?

wangludong, 2020-09-03:
Hi

I deleted most contents of a existed scene and tried to import my fbx and opendrive generated by RoadRunner,
but I can't generate routes markers.

Here are my steps:
1.Make a simple segment of road in RoadRunner.
[image]

2.Export fbx and opendrive with Carla option.
[image]

3.Import the fbx into UE4 and replace the xodr file with my own.
[image]

4.Drag a Open Drive Actor and drop it on my road, but no routes generated. I also tried to check "Generate Routes" Option in Details Panel and got no response either.
[image]

I hope someone could give me some suggestions.
Thanks a lot.

thytless, 2020-09-04:
Hello! The "Generate Routes" Option in OpenDriveActor should work, did you also check the "Show Debug" option below?

wangludong, 2020-09-04:
> Hello! The "Generate Routes" Option in OpenDriveActor should work, did you also check the "Show Debug" option below?

Hi
"Show Debug" option is checked in my project.
Actually, I followed this document [https://www.mathworks.com/help/roadrunner/ug/Exporting-to-CARLA.html](url).
Now I can run PythonAPI\examples\manual_control.py to drive the car manually or automatically.
But when I launched PythonAPI\examples\automatic_control.py, I got a number of errors below.
[image]

It seems like that the simulator can not localize the lane.
I totally don't know whether my operations are right.I think the route mark is supposed to be displayed in order to indicate where the route is.

thytless, 2020-09-04:
Looks like the API had some trouble parsing your opendrive file, maybe find some tools to check it? Also, a closed map is suggested otherwise vehicles may fall out of the map. 
Version 0.9.9 have a better support for opendrive, check it~

wangludong, 2020-09-09:
I found the reason but a little weird.
My previous map had no junctions, so I add some junctions this time and it works.

closed 2020-09-09 by wangludong